sebanc/brunch #1052: How to unlock rootfs?

peteallelite, 2021-05-14:
I need to use an android root app. How do I enable developer options or get onto the developer image firstly?
And secondly how do I gain root access?
I cannot seem to enter "Recovery" mode...

peteallelite, 2021-05-16:
Wow no response. Ok then

zainullahk, 2021-05-16:
Wow your trying to use android developer images you cant there isnt a recovery mode what do you mean the only current way to root chromeos is this https://github.com/CerisWhite/CrAnberry

andiweli, 2023-07-06:
Hmm...

chronos@localhost /usr/local/bin $ curl https://raw.githubusercontent.com/CerisWhite/CrAnberry/main/cranberry.sh | sudo bash

No android system image present in /opt/google/containers/android/