sebanc/brunch #1053: Play store not working after using android_init_fix

alessandrogian, 2021-05-14:
I have just installed rammus image with the latest stable version of Brunch, the play store takes forever to load and loaded only once in debug mode
[image]

princepjamie06832, 2021-05-15:
What recovery are using and your computer specifications?

alessandrogian, 2021-05-15:
HP 280 G1
Intel® Pentium® G3250 with Intel HD Graphics (3.2 GHz, 3 MB cache, 2 cores)
4 GB DDR3-1600 UDIMM (1 x 4 GB)

It works after I wait a bit on the login screen, as usual, to let the android container initialize, however, it still takes a long time to start, is it also possible to enable automatic updates and will it clear the user data?

princepjamie06832, 2021-05-15:
Chrosini runs Linux virtual machine. When select 'Crostini', it'll jump to 'Developers' section to get started set up Linux virtual machine. Enabling updates will not clear your data. Are you using hard drive or SSD? Using mechanical hard drive are taking forever to load and it's not recommended.

alessandrogian, 2021-05-15:
Unfortunately, I'm using an HDD, the problem is that when I click on Linux it doesn't take me to the developers' section, it just doesn't let me click it