sebanc/brunch #1055: Wifi Not Working In Brunch r8125-9.005.01

Mohammad1111Rahat, 2021-05-16:
I Have Installed Crome OS Using Brunch r8125-9.005.01 In Dell Insperion 14 300 series it has core i3 4005u CPU and Wi-Fi card as Dell wireless 1704.802.11b/g/n (2.4) Realtek Wi-Fi driver. Everything works fine accept this Wi-Fi card. I can't Connect to wifi and I didn't Find Any solution for this thats why I have came here I request you please give me a solution for this.

SMARTSKA97, 2021-05-20:
Just check which realtek driver is of ypur system???
Just enable using commands... in readme.md

asukumar10248, 2021-05-24:
"iwlwifi_backport": enable this option if your intel wireless card is not supported natively in the kernel,
try this

Buzzcut22, 2021-07-09:
sir how to use this iwlwifi_backport command?

Please reply with a  step by step tutorial and screenshots to follow