sebanc/brunch #1064: Not Support laptop with 32-bit UEFI which has 64-bit processor

Rajangupta09, 2021-05-20:
boot menu doesnot display the sd card because  it has a 64bit uefi file.

simplyrajatgupta, 2021-05-31:
This will give you support
[https://github.com/sebanc/brunch/issues/1088]()
And our name also matches
Rajat Gupta ~Rajan Gupta
With love from India

Rajangupta09, 2021-06-01:
@simplyrajatgupta I dont think you are understanding my problem

I have 2 laptop,
one asus laptop which has Linux installed on it 
and another one is a  **hp 2-in-1**  laptop it has windows 10 installed on it, and 2gb ram and 32gb rom,
 I want to install chrome OS as my primary OS on this hp laptop.  but I dont want to boot linux on this hp laptop because I have another laptop with linux, and I have tried all other method but the usb does not shows up on the boot menu because this hp laptop has a **32 bit uefi and 64 bit processor**
 
 please help me install it.
 thank you

Sayor0630, 2021-06-02:
What is your laptop's full name?

Rajangupta09, 2021-06-02:
hp x2 210, it has 2gb ram and 32gb storage, with intel atom 64 bit processor

dimasirfandanu, 2021-06-02:
maybe your SD card is the culprit, try installing brunch to laptop's internal disk. I have same setup with you and brunch booted flawlessly. Brunch uses GRUB as the bootloader and should allow cross architecture booting.

simplyrajatgupta, 2021-06-03:
It worked or not?
@Rajangupta09  Please reply

Rajangupta09, 2021-06-03:
@simplyrajatgupta 
[image]
I am getting this error now

Rajangupta09, 2021-06-03:
> maybe your SD card is the culprit, try installing brunch to laptop's internal disk. I have same setup with you and brunch booted flawlessly. Brunch uses GRUB as the bootloader and should allow cross architecture booting.

I cannot because this laptop has 32 gb total storage in which 28gb is available and 15 gb is taken by windows it self .

simplyrajatgupta, 2021-06-04:
> I cannot because this laptop has 32 gb total storage in which 28gb is available and 15 gb is taken by windows it self .

@Rajangupta09 First of all I did not think than an laptop can have an storage just 32gb  and Chrome OS need at least 14 gb and 1 gb for file system and other system information the I suggest that you install just chrome is on that. By following sebanc single boot method.

Sayor0630, 2021-06-10:
> @simplyrajatgupta
> [image]
> I am getting this error now

Isn't this a problem with secure boot? Turn off your secure boot. And then try again.

simplyrajatgupta, 2021-06-10:
> > @simplyrajatgupta
> > [image]
> > I am getting this error now
> 
> Isn't this a problem with secure boot? Turn off your secure boot. And then try again.

It's not my problem but @Rajangupta09 have problem.
I have done that successfully

Sayor0630, 2021-06-10:
> @Sayor0630 It's not my problem but @Rajangupta09 have problem.
> I have done that successfully

Yeah. But he needs help

simplyrajatgupta, 2021-06-10:
>  > @Sayor0630 It's not my problem but @Rajangupta09 have problem.
> > I have done that successfully
> 
> Yeah. But he needs help

I tried to contact him here but he is not replying.

Rajangupta09, 2021-06-17:
> > @simplyrajatgupta
> > [image]
> > I am getting this error now
> 
> Isn't this a problem with secure boot? Turn off your secure boot. And then try again.

No its not, I have turned of secure boot still have this issue

simplyrajatgupta, 2021-06-17:
@Rajangupta09 Single boot Chrome OS on that and this is the last option for 32gb laptop

1. Full install Chrome OS on an usb (from the laptop in which you installed Linux)
2. In the 32gb laptop boot from that usb
3. Open Linux terminal in chrome os (Setting---> Developer--->Linux Beta)
4. And follow brunch wiki guide of installing Chrome OS from usb.

closed 2026-02-26 by github-actions[bot]