sebanc/brunch #1065: Chrome OS Reboot while using Android apps

mnmn565, 2021-05-21:
I have successfully installed Chrome OS on my Asus Transformer Mini T102H, (x5-z8300 processor), whenever I start using an android app, the entire OS crashes within 10 minutes and forces a reboot. It doesn't matter which app I use, whether it's YouTube or Oldschool RuneScape, eventually crashes the system.

asukumar10248, 2021-05-21:
Those are not options, just add them on the kernel command line after "cros_debug" and before "options=...." if any:

"enforce_hyperthreading=1": improve performance by disabling a ChromeOS security feature and forcing hyperthreading everywhere 
this will help you

mnmn565, 2021-05-21:
Hi, I'm not sure I follow what you're trying to say, can you be a bit more clear on how that is related to my system crashing with Android apps?

Thank you.

mnmn565, 2021-05-21:
Well I did as you said and added "enforce_hyperthreading=1" in the kernel command line and no crashes so far, may have solved the problem, I will report here if any further issues arise. Thank you again.

mnmn565, 2021-05-21:
Okay so the system lasted a lot longer compared to before but it did crash again. This time it crashed while I was browsing the Google Play Store.

asukumar10248, 2021-05-22:
"android_init_fix": alternative init to support devices on which the android container fails to start with the standard init, 
try this then give the result

sebanc, 2021-05-22:
I think that's the long lasting linux kernel issue with baytrail devices, try adding "intel_idle.max_cstate=1" or "intel_idle.max_cstate=0" on the kernel command line.

I don't know if this was ever fixed but you could also try kernel 5.10 if you have not already.

mnmn565, 2021-05-22:
Haven't had a crash so far after switching to kernel 5.10. I will keep you updated if anything happens, cheers!

closed 2026-02-26 by github-actions[bot]