sebanc/brunch #1066: Restart in an infinite loop.

XinuGod, 2021-05-22:
Hello, I installed the brunch framework supported by bios boot, because my old computer is too old to support UEFI boot. After the installation is complete, I can boot normally and display the booting kertain, but after a period of time, it will automatically restart in an infinite loop. I don't know why.
[image]

asukumar10248, 2021-05-22:
turn the laptop off then turn on and click R in grub menu

ghot102, 2021-06-03:
the same CPU, you solved it?

SakuragiHa, 2021-06-17:
I have the same exact problem: my computer cannot support uefi, then installed chromeos with mbr_support.tar.gz patch, then it can boot into the stage in the photon then reboot. please help

palaboy, 2021-08-03:
same here successfully installed with mbr patch but after "Booting the kernel." message laptop automatically restart any help to fix this problem?

lentas1, 2022-07-19:
e300 same problem

palaboy, 2022-07-19:
> e300 same problem

Try ChromeOS Flex

closed 2026-02-25 by github-actions[bot]