sebanc/brunch #1067: Download error

LCY000, 2021-05-22:
I don't know what happen!
I already tried to download some different versions.
Who can help me?
[image]

kushmittal2009, 2021-05-22:
This is a very common error. 

There are two steps to this. 
1.) open chrome settings chrome://settings/     Then search safe browsing.  Make it to no protection. This will let you download it. 

2.) Then if it still blocks it you will have to temporarily disable your antivirus if you have one.

LCY000, 2021-05-23:
Thank You!!   I got it!

princepjamie06832, 2021-05-23:
I didn't face any errors when downloading. Nothing happens to anti-virus & safe browsing in web browsers.