sebanc/brunch #1069: Fix Chrome OS on PC Installation, Apps looping

JamesMaria, 2021-05-24:
Apps are looping whenever I try to open Play Store or an Android app, any solutions. I am using a 500 GB HDD, Intel i5-4200u

asukumar10248, 2021-05-24:
turn your computer off then turn it on and wait for 1 minute at login window then type passward then wait for 30 sec now it will work because It is recommended to install Chrome OS on an EXT4 partition on an SSD (solid state drives) for system stability. Support MAY NOT be provided if Chrome OS is run on a hard drive (HDD) or a thumb drive (USB).
Reason: Chrome OS is designed to only run on SSDs or NVMes. Many critical features (Android apps for example) will BREAK when executing this software on a portable USB or an HDD (hard disk drive) or any slower storage technologies, as you will generally log in before such features finish preloading.

asukumar10248, 2021-05-24:
if problem persist then free to ask

JamesMaria, 2021-05-24:
What can I do if choose not to wait for a minute?

asukumar10248, 2021-05-24:
upgrade your laptop from hdd to ssd

khanhtranngoccva, 2021-08-30:
eh, did I write that BASH script? uh xD