sebanc/brunch #1070: Play Store Not Loading

JFer20, 2021-05-24:
-Laptop Specs-
Surface Laptop Go 
Intel UHD Graphics
8 Gb Ram
10th Gen Intel Core i5 processor – 1035G1

I'm currently running Brunch v90 Stable as well as Rammus recovery image v90. 
It seems to be working great, other than the play store not cooperating. I'd seen somewhere about a potential fix "Android_Init" command. I don't even know where to start. Any ideas would be greatly appreciated.

princepjamie06832, 2021-05-24:
'Volteer' recovery image should work Google Play Store. When you got error encountered with code 8, this means don't work that you're selected this recovery image. Mine works without issues and still based on Android Pie. You'll no longer needs "android_init_fix" while on 'Volteer' recovery image.

JFer20, 2021-05-24:
I will try that image now, just checking before i do, my error code is 7, this fix should still apply?

princepjamie06832, 2021-05-24:
You can try 'Volteer', but you'll need to reinstall Chrome OS.

JFer20, 2021-05-24:
works perfect, thanks so much for your help!

princepjamie06832, 2021-05-24:
No problem! This can work perfectly. If there's any problems, let us know.

JFer20, 2021-07-06:
Hello, Recently Volteer was removed from Cros, Do you know of any other recovery image that could still be used with 10th gen Intel?

antoninchadima, 2021-07-07:
Does anyone have a Volteer recovery backup? Would be happy if you could share the image...

antoninchadima, 2021-07-07:
> Hello, Recently Volteer was removed from Cros, Do you know of any other recovery image that could still be used with 10th gen Intel?

Hatch

JFer20, 2021-07-07:
Try this link:

https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_13904.55.0_volteer_recovery_stable-channel_mp-v6.bin.zip

antoninchadima, 2021-07-07:
90: https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_13816.82.0_volteer_recovery_stable-channel_mp-v6.bin.zip
91: https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_13904.55.0_volteer_recovery_stable-channel_mp-v6.bin.zip

pierro78, 2021-11-08:
hello @JFer20 does "2 fingers scrolling" always work fine with your surface laptop go touchpad ??
it sometimes has issues with mine (here is my setup : https://www.reddit.com/r/SurfaceLaptop/comments/qlwaem/chromeos_brunch_seems_to_work_great_on_surface/ : brunch-unstable 20211104 , volteer (chromeos recovery) v94, kernel v5.4 ...)

i-depth, 2023-06-03:
hi, I have the same problem with google play and error code 7 
my laptop has i5-4200u 
4gb ram 
i tried the two image and still the same problem 
i install it on the ssd