sebanc/brunch #1079: Failed to install chrome os on asus VivoBook Flip 14 TP470 notebook

wz6789, 2021-05-27:
1. I have successfully installed chrome os on amd4700cpu laptop. The operation is also normal, using zork.
2. Then the installation of chrome os on asus VivoBook Flip 14 TP470 (cpu: i7-1165G7) failed.
3. The volteer used on the VivoBook Flip 14 TP470 notebook.
4. Error message: Failed during the brunch boot process, it prompts "Cannot find the chrome os file".

asukumar10248, 2021-05-27:
use hatch recovery

wz6789, 2021-05-27:
Okay, I'll try it now. Thank you

wz6789, 2021-05-27:
The problem has not changed. Still prompted: brunch: chromeOS was not found on any device, falling back to shell

wz6789, 2021-05-27:
Is there any alternative. Thank you

princepjamie06832, 2021-05-27:
Use 'Volteer' that will support for 10th-11th Gen and required change kernel to 5.10 before booting into Chrome OS. Change from "/kernel" to "/kernel-5.10" by using nano text editing.

wz6789, 2021-05-27:
Kernel-5.10 is currently used

wz6789, 2021-05-30:
To solve the problem, the pci-e hard disk could not be found because of lack of vmd support. Solved by using the new version of the framework.

closed 2021-05-30 by wz6789