sebanc/brunch #1081: help installing vbox drivers on chromeos

sainAk, 2021-05-27:
I installed chrome os 90 on vbox and just like everyone else I got a blank screen after chromeos logo,
when I tried accessing the tty it was still a blank screen but i was able to see some text for a brief moment so i recorded the screen and played the video frame by frame and I saw this.
[image]

I was able to log in as root so no if we just install the graphics driver we might get it to work...

princepjamie06832, 2021-05-28:
Can't run on virtual machine and only run on physical hardware.

newbit1, 2021-05-28:
> Can't run on virtual machine and only run on physical hardware.

What about the [chrome-os-0.4.22.8](https://archive.org/download/chromeos-0.4.22.8) [Version](https://www.cultofmac.com/22551/how-to-install-googles-chrome-os-on-mac-using-parallels-desktop-5/)? It does run on VMs and it has quite a lot of drivers
in `2/usr/lib/dri/`
```
2413360 Jul 21  2009 i810_dri.so
2700240 Jul 21  2009 i915_dri.so
2778832 Jul 21  2009 i965_dri.so
2482964 Jul 21  2009 mach64_dri.so
2519920 Jul 21  2009 mga_dri.so
2401040 Jul 21  2009 r128_dri.so
2486384 Jul 21  2009 r200_dri.so
2499600 Jul 21  2009 r300_dri.so
2455572 Jul 21  2009 radeon_dri.so
2388688 Jul 21  2009 s3v_dri.so
2454804 Jul 21  2009 savage_dri.so
2478960 Jul 21  2009 sis_dri.so
2240948 Jul 21  2009 swrast_dri.so
2470704 Jul 21  2009 tdfx_dri.so
2333140 Jul 21  2009 trident_dri.so
2405176 Jul 21  2009 unichrome_dri.so
```
There must be a way to recompile and insert them into the outer Linux ENV from ChromeOS?!
Every Linux Live CD/USB Installation has them as well.

sainAk, 2021-05-28:
> What about the [chrome-os-0.4.22.8](https://archive.org/download/chromeos-0.4.22.8) [Version](https://www.cultofmac.com/22551/how-to-install-googles-chrome-os-on-mac-using-parallels-desktop-5/)? It [...]

would it be possible to install these on arunning instance
I don't lnow about gentoo, and the confusing filesystem of chromeos os making it more difficult for me to modiy anything
also I am able to run chrome os, its taking inputs I'm just not able to see what I am typing,
I looking at init messeges I could see that it was detecting the vbox network card so network shouldn't be an issue either so i'm still trying to get ssh to work on it.

newbit1, 2021-05-28:
> Can't run on virtual machine and only run on physical hardware.
> would it be possible to install these on arunning instance

I tried already both, the DRIs from the old chromeos and a modern linux.
Can't tell if it really changed something, issue remains.
> If possible can you upload these drivers somewhere.

[dri.zip](https://github.com/sebanc/brunch/files/6559787/dri.zip)

I am not yet so familiar with this thing of adding a driver into a linux system.
Could need some basic info to read about it. But I guess it is not that difficult.
I assume you have to tell the kernel at least, that the drivers are there.
I did find something about it though, but they are talking about the Android level,
not the Linux level.
[Replacing Android drivers on stock Android.](https://gallium.readthedocs.io/en/latest/android.html)
As you can see in Issue [#303](https://github.com/sebanc/brunch/issues/303)
He (sebanc), knows already how to implement them, but doesn't have the proper ones.
So, If I could know how to implement them, I would try to make/find proper ones.
> 
> I don't know about gentoo, and the confusing filesystem of chrome os is making it more difficult for me to modiy anything
> also I am able to run chrome os, its taking inputs but the display is not worling so i'm not able to see what I am typing.
> Looking at init messeges I could see that it was detecting the vbox network card so network shouldn't be an issue so i'm still trying to get ssh to work on it.

Wasn't there a way to boot straight into a linux shell? instead of trying to start a display server?

sainAk, 2021-05-28:
I ran into some issues switching back and forth between the VTs and saw this 
[image]
The drivers are already present in the kernel and frecon is using them(I thing due to this i am able to see flickers of text) but its still stuggling to display anything
if I boot using VboxSVGA drivers it doesnt detect any display at all and I get stuck on the dmesg screen without a shell prompt

TheAlphaReturns, 2022-05-02:
Perhaps steal the drivers from chromiumOS / cloud ready / etc.