sebanc/brunch #1083: Could it boot in a partition formatted as Btrfs?

1457384613gh, 2021-05-28:
Could it boot in a partition formatted as Btrfs?
If it could boot in a partition formatted as Btrfs and make good use of function named subvolume, it would be superior and similar to Mac OS booting in APFS.

simplyrajatgupta, 2021-05-31:
No, Only NTFS and ext4 supported
Check this also may be helpful
[https://github.com/sebanc/brunch/issues/1088]()

1457384613gh, 2021-05-31:
> No, Only NTFS and ext4 supported
> Check this also may be helpful
> https://github.com/sebanc/brunch/issues/1088

I'm sure it can be, because I have managed !

simplyrajatgupta, 2021-06-01:
> > No, Only NTFS and ext4 supported
> > Check this also may be helpful
> > https://github.com/sebanc/brunch/issues/1088
> 
> I'm sure it can be, because I have managed !

But NTFS and ext4 is recommended by @sebanc

simplyrajatgupta, 2021-06-14:
Please close your issue

simplyrajatgupta, 2021-06-14:
@1457384613gh  Please close your issue

closed 2022-05-17 by 1457384613gh