sebanc/brunch #1084: How to Install ChromeOS from Fedora ?

1457384613gh, 2021-05-28:
Fedora cannot `apt-get install cgpt.deb`.
Even though I `alien -r cgpt.deb`, and `yum -y localinstall cgpt.rpm`, it fail to install.
Could you edit a brand new *chromeos-install.sh* for Linux distribution with both rpm and yum？
Maybe it would bettter install in a partition formatted as Btrfs, and Root-A, OEM, stage and so on could be sub-volumes, which is friendly to SSD.

WiseMetalhead, 2021-05-30:
You can install cgpt via the command:
# sudo dnf install vboot-utils
Works also on Fedora 34.

1457384613gh, 2021-05-30:
> You can install cgpt via the command:
> **sudo dnf install vboot-utils**
> Works also on Fedora 34.

Thanks for your reply and how earnest you are !

1457384613gh, 2021-05-30:
Wish the next right thing might be to merge EFI system partitions and make use of function called subvolume.

1457384613gh, 2022-03-09:
> > > You can install cgpt via the command:
> > > **sudo dnf install vboot-utils**
> > > Works also on Fedora 34.
> > 
> > 
> > Thanks for your reply and how earnest you are ! One installed into Btrfs runs even fast than another installed into ext4.
> 
> did it work? were you able to sucessfully install chromeos using fedora

using fedora √
btrfs ？
I don't know if R-98 …

closed 2022-05-17 by 1457384613gh