sebanc/brunch #1085: VMD Support

ron20hg, 2021-05-29:
I tried install brunch r90 with volteer90 on my zenbook 14".
It has an NVMe drive and after research found that it also needs VMD driver support in initramfs.

How to add the vmd to the brunch initramfs?

wz6789, 2021-05-29:
I may have encountered your problem. Did you solve it?

ron20hg, 2021-05-29:
No. I don't understand how to change the initramfs in brunch. The only lead I have is to recompile the 5.10 kernel but I'm not ready for this yet.

sebanc, 2021-05-29:
That's interesting, I had no idea that this driver was necessary for some nvme drives, I will enable it in my next build and post the link here.

sebanc, 2021-05-30:
Could you try the below release ? It has VMD enabled
https://github.com/sebanc/brunch-unstable/releases

wz6789, 2021-05-30:
The hard disk can be found, but it freezes after entering the chromeos logo.

wz6789, 2021-05-30:
[image]
Stuck here

wz6789, 2021-05-30:
The installation was successful. Testing compatibility.

wz6789, 2021-05-30:
Compatibility issues that have been found include:
 1. The tablet mode cannot be activated by rotating the screen. 
 2. The microphone cannot be used.
 3.  3. The display brightness cannot be adjusted. 

The computer model is: ausu zenbook14 (cpu: i7-1165g7)

ron20hg, 2021-05-30:
I'm using the Zenbook 14" (no Flip).
And I've managed to get ChromeOS working! with the unstable version of r90 volteer
Thanks @sebanc!

My brightness control is working, mic not working yet.... (not given up)
Camera works with 'no_camera_config' option, also power button menu with 'acpi_power_button'.