sebanc/brunch #1087: Chrome OS is not opening after installation

TechnicalExpo, 2021-05-30:
I installed Chrome OS in my PC as Multi Boot and Single.... But in Both cases it's not opening everything was good Installation but when I started system it always stuck at Chrome Logo screen.. System nkt opens.

I have ASUS and Processor is Intel core i5
I tried rammus, zork and samus.. But in each I faced same problem it always stuck there....

On this page it stuck. https://images.app.goo.gl/dBAY1YkqfNua5Q5K6 

If anyone can help me so please let me know...

asukumar10248, 2021-05-31:
you have intel graphics or nividia graphics

TechnicalExpo, 2021-05-31:
> you have intel graphics or nividia graphics

Nvidia graphics

asukumar10248, 2021-05-31:
nividia graphics is not yet supported that why this problem is comeing
to solve this problem 
1 turn on your computer
2 when the grub menu apper click e before it start booting
3 then add module_blacklist=nouveau after cros debug and option 
4 then save changes and you will be greated to chrome os after roofs is build and applied

TechnicalExpo, 2021-05-31:
> nividia graphics is not yet supported that why this problem is comeing
> to solve this problem
> 1 turn on your computer
> 2 when the grub menu apper click e before it start booting
> 3 then add [...]

Mean after doing thus OS will open...?

TechnicalExpo, 2021-05-31:
Which one is good img... Rammus, samus, zork for my PC.
Its ASUS, Processor is Intel core i5, graphic is nvidia... 

and you told me to edit and type that.... Will it work...

asukumar10248, 2021-05-31:
> Which one is good img... Rammus, samus, zork for my PC.
> Its ASUS, Processor is Intel core i5, graphic is nvidia...
> 
> and you told me to edit and type that.... Will it work...

use rammus 
yes it will work

simplyrajatgupta, 2021-05-31:
Due to several errors it may happen
But not after this
[https://github.com/sebanc/brunch/issues/1088]()

TechnicalExpo, 2021-05-31:
> nividia graphics is not yet supported that why this problem is comeing
> to solve this problem
> 1 turn on your computer
> 2 when the grub menu apper click e before it start booting
> 3 then add [...]

How to solve it... After editing

TechnicalExpo, 2021-05-31:
> Due to several errors it may happen
> But not after this
> https://github.com/sebanc/brunch/issues/1088

When I press ctrl + x it boot.. But there I found error = no such module...
Press any key to continue.... 
After that Brunch load and it freeze there than not loading chrome logo... mean remains on the "Loading Brunch Framework..." not going ahead....

simplyrajatgupta, 2021-06-01:
> When I press ctrl + x it boot.. But there I found error = no such module...
> Press any key to continue....
> After that Brunch load and it freeze there than not loading chrome logo... mean remains on [...]

Did you matched the release no. And recovery img no. 
Ex- recovery img no 90 = brunch r90

simplyrajatgupta, 2021-06-03:
@TechnicalExpo  did you solved your problem

lucasgabmoreno, 2023-11-22:
Same problem, and solved changing fast boot as "disabled" in Windows and BIOS