sebanc/brunch #1088: Dual boot Chrome OS with Windows without any error at all. [By creating chromos.img]

simplyrajatgupta, 2021-05-31:
Hello!
Dear, Dearer, Dearest ladies and gentleman of curious world who want to try Chrome OS on their laptop running (bios legacy / uefi) and with laptop as old as having 2 gb ram.

First of all, Please don't go back without reading this fully because of many error you encounter during installation process.
And I am an Indian so i will know better how to use **जुगाड/ an idea to solve any problem with very ease**

So let's start

1. **Please don't go back because of hearing this step because I will not use any of method you know** Create a Linux bootable pen drive ( any distribution not only Linux mint ) 
guide here [https://www.howtogeek.com/howto/linux/create-a-bootable-ubuntu-usb-flash-drive-the-easy-way/]()

2. Go to [https://cros.tech]() and click on the three line button at top left corner and select **table** now find your recommended image 

rammus for intel processor 
samus if you have laptop older than 2014
grunt for amd 
**you can also find more useful information on sebanc/brunch readme.md file**

3. go to sebanc/ brunch on github and download the release according to the chrome os recovery image number **for example r90 for recovery image number 90 **

4. extract all the downloaded files

5. copy the extracted files to the usb drive

2. Restart your pc

3. Press your boot hot key 
 list here [https://www.disk-image.com/faq-bootmenu.htm ]()

4. Select your usb drive name.

5. Take a hammer and press the **enter** key 🤣😂
Joke here [https://images.app.goo.gl/vQdHPYqWm7BPVKh4A]()

**Ok jokes aside ** lets continue

6. Now open **Gparted ** 

7. Select the drive in which you store your data [If you are having a single** c** drive then select it ] **One more thing if you doesn't have gparted then open a terminal and type **
```
sudo apt update 
```
and press enter then after completion type 
```
sudo apt-get install gparted
```
If asked for (Y/n) then type **Y**

8. Now, Right click on it and select **RESIZE/MOVE** OPTION

9. In the box labeled with **Free space following (mib)** type 
_internal storage you want in chrome os in mb + 11264_

10. click on the three boxes one by one 

11. Click on **Resize/Move** button

12. Now right click on the new unallocated space and select **New** option **identify by the disk size** [ one more problem may occur here comment below by writing the step number]

13. Now in the file system box select **ntfs** and click on the **add** button.

14. Click on the **green tick button **

[image]

15. open files app

16. go to the **file system** folder

17 select **cdrom** folder

18. here you will find your five files which you have added to the usb drive [if you have created an folder then you will find that also here if it is true then go inside the folder]

19. rename the `chrome os recovery image ending with .bin` to  **chromeos.bin** _please check spelling or copy paste from here_

19. right click on an empty area and select **open in terminal**

20. type the following command one by one 

- sudo apt-get update
-  sudo apt-get install figlet 
- sudo apt-get install pv 
- sudo apt-get install cgpt

21. then the most important step come here

type the command 

```
sudo bash chromeos-install.sh -src chromeos.img -dst 
```
do not press enter

22. open **disks** app 

23. select the partition in which you have to create the disk image **identify by size**

24. click on play button **you may get an error here if yes the comment below by typing step number **
[image]

25. now copy the partition mount point 
[image]

26. go back to the terminal and right click and select paste

it may look like this 
**sudo bash chromeos-install.sh -src chromeos.img -dst /media/123/Q34WE65WE55EW** **DO NOT COPY MY ONE**

27. ADD A SLASH AT THE END 
AFTER THAT IT MAY LOOK LIKE THIS

**sudo bash chromeos-install.sh -src chromeos.img -dst /media/123/Q34WE65WE55EW/** PLEASE NOTICE AT THE END

28.NOW TYPE THE NAME YOU WANT TO GIVE TO YOUR CHROME OS IMAGE FILE 
FOE EX-  **chromeos.img**

AFTER THAT IT MAY LOOK LIKE THIS

**sudo bash chromeos-install.sh -src chromeos.img -dst /media/123/Q34WE65WE55EW/chromeos.img**

29. open **disks** 
 and look at the size of the partition in which you want your **chromeos.img**

if it is in point then take the value without point 
for ex 20.6 ~ 20
           50.1 ~ 50
           61.7 ~ 61

after that subtract 1 from the value
for ex 20 -1 =19
           50-1 = 49
           61-1 = 60
    (now eat almonds and remember this value for five sec 🧠)

29. now give size to your chromeos.img 
 by typing **-s** and size in gb 

for ex  -s 19
            -s 49
            -s 60

at the end your command should look like this 

**sudo bash chromeos-install.sh -src chromeos.img -dst /media/123/Q34WE65WE55EW/chromeos.img -s 60**

30. this time seriously

ameyjoshi3068, 2021-06-02:
I really appreciate your efforts.
I think this guide is worth suggesting to add in **Wiki**. But still we cannot neglect some things(Too long, region specific language.
I will try to correct them and add an pull request. You are free to comment there. Sebanc will take final decision of this.

simplyrajatgupta, 2021-06-02:
> will you please copy and paste your guide here in edit mode?
> This will be helpful for me (I am using same images as yours)

    I don't know how to do that but if you say then I can create a pdf and send you

ameyjoshi3068, 2021-06-02:
Sorry it was my mistake.
just edit your comment and copy everything inside `write` box.
paste it [here](https://pastebin.ubuntu.com/) and share the txt

simplyrajatgupta, 2021-06-02:
> Sorry it was my mistake.
> just edit your comment and copy everything inside `write` box.
> paste it [here](https://pastebin.ubuntu.com/) and share the txt

https://pastebin.ubuntu.com/p/mY4Bbw5QFf/

ameyjoshi3068, 2021-06-02:
Thank you so much. I really like your writing style.

simplyrajatgupta, 2021-06-02:
I got it in the chapter of adjective and everyone love it!
Dear Dearer Dearest @ameyjoshi3068

ameyjoshi3068, 2021-06-02:
**Hello**!
Dear, Dearer, Dearest ladies and gentleman of curious world who want to try Chrome OS on their laptop running (bios legacy / uefi) 64 bit (X64) pc or laptop.It definitely runs a lot faster than windows even with low RAM.

So let's start

**Downloads**
Download these files first

1.Linux Mint ISO file [Select Cinnamon](https://linuxmint.com/download.php)

2. Go to [This Website](https://cros.tech) and click on the three line button at top left corner and select **table** now find your recommended image and download it.

rammus for intel processor 
samus if you have laptop older than 2014
Zork  for AMD Ryzen
**you can also find more useful information on [Main Page](https://github.com/sebanc/brunch#readme)**

3. go to [Releases](https://github.com/sebanc/brunch/releases) and download the latest release.

**Procedure**

1. create a new folder "Chrome OS" and extract file no 3 there.

2. extract file no 2 in the same folder and rename it as `recovery.bin`. 

3. Now, folder Chrome OS should contain these files

[image]

4. create live USB of Linux mint. guide [here](https://www.howtogeek.com/howto/linux/create-a-bootable-ubuntu-usb-flash-drive-the-easy-way/)

5. after USB Device is created, copy "Chrome OS" folder in your it.

6. Insert USB and restart your PC.

7. Press your boot hot key 
 [list here](https://www.disk-image.com/faq-bootmenu.htm)

8. Select your usb drive name.

9. Take a hammer and press the **enter** key 🤣😂
 [Joke here](https://images.app.goo.gl/vQdHPYqWm7BPVKh4A)

**Ok jokes aside ** lets continue

10. Make sure you are connected to internet. press **Ctrl+Alt+t**
11. Paste the following code in terminal

> `sudo apt update`
> `sudo apt install gparted pv cgpt figlet -y

12. Open **Gparted** from menu.

13. Select the drive in which you store your data [If you are having a single** c** drive then select it ]

14. Now, Right click on it and select **RESIZE/MOVE** OPTION

15. In the box labeled with **Free space following (mib)** type 
`internal storage you want in chrome os in mb + 11264`

[GB to MB Converter](http://www.kylesconverter.com/data-storage/gigabytes-(binary)-to-megabytes-(binary))

> For example, I want 32 GB internal storage, I will type `44032` in `Free space following (mib)` box

*NOTE*:- now eat almonds and remember this number 🧠 
Just Kidding, Write it somewhere.

16. Click on **Resize/Move** button

17. Now right click on the new unallocated space and select **New** option.
[... 108 lines cut ...]
[image]
[image]

53. Go to the partition containing .txt file and copy the code

54. Come back to the notepad opened by grub2win and paste the code
[image]

55. Save the file by pressing Ctrl+S

56. [image]
[image]
[image]
[image]

57. [image]
[image]

58. Get some Pepsi and show your swag with chrome os. 
[image]
[image]

Thanks,
With love ❤️ Team brunch and @simplyrajatgupta 
#HappyChromeOSjourney

ameyjoshi3068, 2021-06-02:
I have edited upto ++++++++++++ line.
If you have other suggestions or improvements, please comment. I will add it tommorow.

simplyrajatgupta, 2021-06-02:
> I have edited upto ++++++++++++ line.
> If you have other suggestions or improvements, please comment. I will add it tommorow.

Is it coming to readme.md
I am very excited

simplyrajatgupta, 2021-06-02:
In step 40 img is missing

simplyrajatgupta, 2021-06-02:
In step 40: img is missing
And in line 2 at last **~~is~~** id
Step 52: remove the first img **edit custom code img** it should not be there at last it should be.
In step 56: One more img is missing

simplyrajatgupta, 2021-06-02:
In step 1 and 2 where is **file no 1**
In step 11 write the code as bulleted list or write there Paste..... terminal one by one
In step 29: The tmpmount command will create the chromeos.img in the same **cdrom** or in the /dev/XXXX

ameyjoshi3068, 2021-06-03:
**Hello**!
Dear, Dearer, Dearest ladies and gentleman of curious world who want to try Chrome OS on their laptop running (bios legacy / uefi) 64 bit (X64) pc or laptop.It definitely runs a lot faster than windows even with low RAM.

So let's start

## **Downloads**
Download these files first

**File 1.** Linux Mint ISO file [Select Cinnamon](https://linuxmint.com/download.php)

**File 2**. Go to [This Website](https://cros.tech) and click on the three line button at top left corner and select **table** now find your recommended image and download it.

Currently:
- "rammus" is the recommended image for devices with Intel CPU up to 9th gen.
- "zork" is the image to use for AMD Ryzen (Just change /kernel with /kernel-5.10 in step 00).
- "grunt" is the image to use for AMD Stoney Ridge.

**you can also find more useful information on [Main Page](https://github.com/sebanc/brunch#readme)**

**File 3**. go to [Releases](https://github.com/sebanc/brunch/releases) and download the latest release.

**File 4**(Optional). Download and install [Grub2Win](https://sourceforge.net/projects/grub2win/) If you want to dual boot with windows.

## **Procedure**

1. create a new folder "Chrome OS" and extract file no 3 there.

2. extract file no 2 in the same folder and rename it as `recovery.bin`. 

3. Now, folder Chrome OS should contain these files

[image]

4. create live USB of Linux mint. guide [here](https://www.howtogeek.com/howto/linux/create-a-bootable-ubuntu-usb-flash-drive-the-easy-way/)

5. after USB Device is created, copy "Chrome OS" folder in your it.

6. Insert USB and restart your PC.

7. Press your boot hot key 
 [list here](https://www.disk-image.com/faq-bootmenu.htm)

8. Select your usb drive name.

9. Take a hammer and press the **enter** key 🤣😂
 [Joke here](https://images.app.goo.gl/vQdHPYqWm7BPVKh4A)

**Ok jokes aside ** lets continue

10. Make sure you are connected to internet. press **Ctrl+Alt+t**
11. Paste the following code in terminal

> `sudo apt update`

> `sudo apt install gparted pv cgpt figlet -y`

12. Open **Gparted** from menu.

13. Select the drive in which you store your data [If you are having a single** c** drive then select it ]

14. Now, Right click on it and select **RESIZE/MOVE** OPTION

15. In the box labeled with **Free space following (mib)** type 
`internal storage you want in chrome os in mb + 11264`

[GB to MB Converter](http://www.kylesconverter.com/data-storage/gigabytes-(binary)-to-megabytes-(binary))

> For example, I want 32 GB internal storage, I will type `44032` in `Free space following (mib)` box

*NOTE*:- now eat almonds and remember this number 🧠 
[... 123 lines cut ...]
[image]

5. Go to Chrome OS folder and open the text file and copy the code

6. Come back to the notepad opened by grub2win and paste the code
[image]

7. Save the file and exit.

[image]
[image]
[image]
[image]

8. Now, Restart your PC. You should see Grub2win.

9. Select Chrome OS with keyboard.

58. Get some Pepsi and show your swag with chrome os. 
[image]
[image]

Thanks,
With love ❤️ Team brunch
#HappyChromeOSjourney

simplyrajatgupta, 2021-06-03:
I mentioned some issues fix them

ameyjoshi3068, 2021-06-03:
Now Its final. Please tell any issues Now

simplyrajatgupta, 2021-06-04:
At some places  bold text is showing like * * for ex at places written apple
Step 29: give a hint to reduce size by one gb also

Setup grub 2 win 
Step 10: Apply & Ok button info is not given

simplyrajatgupta, 2021-06-04:
> Now Its final. Please tell any issues Now

@ameyjoshi3068  did you booted successfully on your laptop?
I am asking this because 
I am getting
error: partition not found
error: you need to load the kernel first

I think it is happening because when I first done that perfectly I was in legacy
But I recently reinstalled win 10 in UEFI.

And while making the img in ubuntu I am getting 
Partition 3: No space left on device
Par.......4: No space left on device
And continued upto Partition 12 and then it gives Chrome OS disk image created.

ameyjoshi3068, 2021-06-04:
I tried installing in UEFI and it worked fine.
I dont have legacy bios PC

ameyjoshi3068, 2021-06-04:
The error you got "No space left on device" seems to be responsible for failed boot.
Have you tried installing with instructions from readme.md and got no error?

simplyrajatgupta, 2021-06-04:
> The error you got "No space left on device" seems to be responsible for failed boot.
> Have you tried installing with instructions from readme.md and got no error?
> 

@ameyjoshi3068 I got the error not during the boot but while creating the disk img in the terminal. So there is no relation with failed boot that is another topic.

simplyrajatgupta, 2021-06-04:
> Is it coming to readme.md
> I am very excited
Please Reply

ameyjoshi3068, 2021-06-04:
Sorry for delayed reply, I was busy in some irl stuff

I will add an pull request to add the link of  this guide in readme.md
And I will also try to add whole guide in wiki.

ameyjoshi3068, 2021-06-04:
> The error you got "No space left on device" seems to be responsible for failed boot.
> Have you tried installing with instructions from readme.md and got no error?

https://github.com/sebanc/brunch/issues/769#issuecomment-753536795
This is not a serious issue.

simplyrajatgupta, 2021-06-05:
@ameyjoshi3068  And what about 
error:no such partition
error:you need to load the kernel first
**During boot.**

ameyjoshi3068, 2021-06-05:
If you have installed Chrome OS on NTFS partition, then try it on ext4.
 cross check your grub configuration, cross check partition number, try recreating chromeos.img

Read this https://github.com/sebanc/brunch/issues/77#issuecomment-627695332

ameyjoshi3068, 2021-06-05:
Just for clarity, I am understanding of this issue during boot is on MBR/ BIOS device

If it is on UEFI device, then the only culprit seems to be your grub configuration or corrupted image

simplyrajatgupta, 2021-06-05:
@ameyjoshi3068  is there any method to add system boot entries to grub

simplyrajatgupta, 2021-06-06:
> @ameyjoshi3068  @sebanc is there any method to add an option to open system boot menu from grub.

@ameyjoshi3068  @sebanc  Please reply

sebanc, 2021-06-12:
Sorry, I have been very busy... Thanks a lot for the detailed guide :) it looks great and is now referred to in the main Readme

supersri811, 2021-06-12:
> @ameyjoshi3068 And what about
> error:no such partition
> error:you need to load the kernel first
> **During boot.**

Even I have the same issue

ameyjoshi3068, 2021-06-12:
@simplyrajatgupta Please close this issue and I will also remove BIOS/MBR part from the guide. Since there are reports that it actually doesn't work.

simplyrajatgupta, 2021-06-13:
> > @ameyjoshi3068 And what about
> > error:no such partition
> > error:you need to load the kernel first
> > **During boot.**
> 
> Even I have the same issue

@supersri811 I am having a solution for that
1. Follow the UEFI guide (carefully without any spelling mistakes)  upto step 31 and after that copy everything written inside the star box.
*****************************
menuentry ...............

.........................../initrarmfs.img
******************************
2. While setting up grub2win 
write the 
menuentry Chromeos class **debug**
............
Code

3. Now remove **rmmod tpm** and **menuentry ........ debug ** line and also  don't forget to remove 
# }
At the end.
4.restart your pc.

skalwaghe-56, 2024-10-02:
> **Hello**! Dear, Dearer, Dearest ladies and gentleman of curious world who want to try Chrome OS on their laptop running UEFI or BIOS Legacy 64 bit (X64) .It definitely runs a lot faster than windows even with low RAM.
> 
> NOTE:- Dual booting is not natively supported with BIOS/MBR devices. If you want to try Chrome OS via full disk install, then view **For BIOS/MBR Devices**
> 
> [Find whether your PC supports UEFI or MBR/BIOS](https://itsfoss.com/check-uefi-or-bios/)
> 
> So let's start
> 
> ## **Downloads**
> Download these files first
> 
> **File 1.** Linux Mint ISO file [Select Cinnamon](https://linuxmint.com/download.php)
> 
> **File 2**. Go to [This Website](https://cros.tech) and click on the three line button at top left corner and select **table** now find your recommended image and download it.
> 
> Currently:
> 
> * "rammus" is the recommended image for devices with Intel CPU up to 9th gen.
> * "zork" is the image to use for AMD Ryzen (Just change /kernel with /kernel-5.10 in step 9 of "grub2WIn).
> * "grunt" is the image to use for AMD Stoney Ridge.
> 
> **you can also find more useful information on [Main Page](https://github.com/sebanc/brunch#readme)**
> 
> **File 3**. go to [Releases](https://github.com/sebanc/brunch/releases) and download the latest release.
> 
> **File 4**(Optional). Download and install [Grub2Win](https://sourceforge.net/projects/grub2win/) If you want to dual boot with windows.
> 
> **File 5**(Only for BIOS/MBR devices) Download and extract mbr_support.tar.gz in your Chrome OS folder.(You can find it on homepage)
> 
> [image]
> 
> # For UEFI devices.
> ## **Procedure**
> 1. create a new folder "Chrome OS" and extract file no 3 there.
> 2. extract file no. 2 in the same folder and rename it as `recovery.bin`.
> 3. Now, folder Chrome OS should contain these files.
> 
> [image]
> 
> 4. create live USB of Linux mint. guide [here](https://www.howtogeek.com/howto/linux/create-a-bootable-ubuntu-usb-flash-drive-the-easy-way/)
> 5. after USB Device is created, copy "Chrome OS" folder in your it.
> 6. Insert USB and restart your PC.
> 7. Press your boot hot key
>    [list here](https://www.disk-image.com/faq-bootmenu.htm)
> 8. Select your usb drive name.
> 9. Take a hammer and press the **enter** key 🤣😂
>    [Joke here](https://images.app.goo.gl/vQdHPYqWm7BPVKh4A)
> 
> **Ok jokes aside** lets continue
> 
> 10. Make sure you are connected to internet. press **Ctrl+Alt+t**
> 11. Paste the following code in terminal
> 
> > `sudo apt update`
> 
> > `sudo apt install gparted pv cgpt figlet -y`
> 
> 12. Open **Gparted** from menu.
> 13. Select the drive in which you store your data [If you are having a single** c** drive then select it ]
> 14. Now, Right click on it and select **RESIZE/MOVE** OPTION
> 15. In the box labeled with **Free space following (mib)** type
>     `internal storage you want in chrome os in mb + 11264`
> 
> [GB to MB Converter](http://www.kylesconverter.com/data-storage/gigabytes-(binary)-to-megabytes-(binary))
> 
> > For example, I want 32 GB internal storage, I will type `44032` in `Free space following (mib)` box
> 
> _NOTE_:- now eat almonds and remember this number 🧠 Just Kidding, Write it somewhere.
> 
> 16. Click on **Resize/Move** button
[... 80 lines cut ...]
> 
> 10. Save the file and exit notepad.
> 11. Click "Apply" and "OK"
> 
> **Note**:- Never press cancel or exit by cross button after exiting notepad. If you do, changes will not be saved.
> 
> 12. Now, Restart your PC. You should see Grub2win.
> 13. Select Chrome OS (with keyboard)(mouse doesn't work)
> 14. Get some Pepsi and show your swag with chrome os.
> 
> [image]
> 
> # For BIOS/MBR devices
> 1. Follow guide mentioned above upto step 11 in "Procedure" section of "For UEFI".
> 2. Extract mbr_support.tar.gz in Chrome OS folder (overwriting files when requested).
> 3. right click on an empty area and select **open in terminal**
> 4. Enter this command
> 
> `sudo bash chromeos-install.sh -src recovery.bin -dst chromeos.img`
> 
> 5. Done. Simply restart your pc and wait. Chrome OS will be ready to use.
> 
> Thanks, With love ❤️ Team brunch #HappyChromeOSjourney

Hey @ameyjoshi3068 I tried this method carried out each and every method properly but when I executed the command to create the ChromeOS image it made many partitions of my drive and messed it all up, I don't know why I created images many time they weren't successful but this never happened. Any suggestions or should I try once more?

skalwaghe-56, 2024-10-03:
Hi booted into brunch at grub2win stuck at chrome os logo