sebanc/brunch #1090: GPD Win3 touchscreen not working

benglim, 2021-06-02:
Managed to get everything working except for touchscreen. Used the volteer recovery image.

Need to update kernel. can you help to compile using tghe guide here?

https://aur.archlinux.org/packages/goodix-gpdwin3-dkms/

sebanc, 2021-06-12:
If you are still interested, you can try the "goodix" option in the latest brunch-unstable release:
https://github.com/sebanc/brunch-unstable/releases

benglim, 2021-06-15:
Thanks. I tried, but it doesn't seem  to work. Maybe need special   handling as in the link above for dkms for GPD WIN3.

from dmesg:
[   24.736652] Goodix-TS i2c-GDIX1002:00: supply AVDD28 not found, using dummy regulator
[   24.736698] Goodix-TS i2c-GDIX1002:00: supply VDDIO not found, using dummy regulator
[   24.737162] udevd[941]: Process '/bin/chown power:power /sys//devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.0/bluetooth/hci0/identity' failed with exit code 1.
[   24.738024] Goodix-TS i2c-GDIX1002:00: ID 911, version: 1060
[   24.743536] Goodix-TS i2c-GDIX1002:00: Invalid config (0, 0, 0), using defaults
[   24.743601] input: Goodix Capacitive TouchScreen as /devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-16/i2c-GDIX1002:00/input/input14

sebanc, 2021-06-18:
Actually that's just what I did, option "goodix" uses the dkms module you pointed.

Could you post a full dmesg ?

benglim, 2021-06-19:
Great!!!! It works. I forgot to put the goodix option earlier.

Much appreciated.

closed 2021-06-19 by benglim