sebanc/brunch #1091: (SB2) Intermittent Touch Lost

jheelme, 2021-06-02:
Hello, I multi-booted using SB2 (GTX 1050) and am using the Branch R90.

When I played the game with all the ipts options enabled, something was weird.

When I tested the touchscreen, I lost touch intermittently on multi-touch of more than 2 points.

If two fingers meet in the same horizontal position, one touch may disappear or both may disappear.

Do you know what the problem is?

kerta1n, 2021-06-03:
Hello, I assume you are referring to the Surface Book 2. NVIDIA graphics are not supported on ChromeOS currently, so your Surface is actually using your Intel graphics, however for the IPTS flag, you may have misconfigured your GRUB, or are using kernel 4.19 which results in error.

jheelme, 2021-06-08:
> Hello, I assume you are referring to the Surface Book 2. NVIDIA graphics are not supported on ChromeOS currently, so your Surface is actually using your Intel graphics, however for the IPTS flag, you may have misconfigured your GRUB, or are using kernel 4.19 which results in error.
I tried to disconnect the tablet and boot it without using Nvidia graphics, and kernel 5.4 is applied by default.

Nevertheless, if you touch the same longitudinal or horizontal axis at the same time, you lose points.

closed 2026-02-25 by github-actions[bot]