sebanc/brunch #1092: Update Brunch Framework: Not Working as Intended?

darethehair, 2021-06-03:
Taking instructions from this location to update my Brunch framework only:

https://github.com/sebanc/brunch#update-only-the-brunch-framework

1) Update command appears to succeed with latest stable framework, but with 'pv' command missing.  I now believe that this is the main problem:

chronos@localhost / $ sudo chromeos-update -f ~/Downloads/brunch_r90_stable_20210523.tar.gz 
rootc.img
rootc.img
/usr/sbin/chromeos-update: line 56: pv: command not found
Brunch updated.

2) Reboot and check Brunch version -- expecting 'r90' or something like that:

chronos@localhost / $ cat /etc/brunch_version 
Brunch r87 20201227

Also notice that this file has not changed since 2020:

chronos@localhost / $ ls -l /etc/brunch_version 
-rw-r--r--. 1 root root 20 Dec 27 03:50 /etc/brunch_version

Have I misunderstood how this is supposed to work?  Or is the 'pv' command missing, and I need somehow to install it?

Thanks!

sebanc, 2021-06-03:
you were supposed to update to brunch r88 before going to chromeos r89 as mentionned in the release notes:
https://github.com/sebanc/brunch/releases/tag/r88-stable-20210223

This issue is not a very complicated to solve though, extract rootc.img from the brunch archive and flash it over chromeos partition 7 using "dd". (if you use dual boot, the chromeos partition is /dev/loop15p7)

darethehair, 2021-06-03:
As an alternative, can the older r88 be installed now, and then I repeat
the current r90 attempt?

If not, would the 'dd' command look like this?

sudo dd if=root.img of=/dev/loop15p7

Looking at just one of my dual-boot installs, there is no '/dev/loop15p7'
partition, just the following:

***@***.*** / $ df | grep loop
/dev/loop15p3
 4128448  2527220          1601228  62% /
/dev/loop15p1
41866332 10173788         29536104  26% /mnt/stateful_partition
/dev/loop15p8
 11760       28            11408   1% /usr/share/oem
/dev/loop1
128      128                0 100% /run/chromeos-config/private
/dev/loop5
 109568   109568                0 100%
/usr/share/chromeos-assets/speech_synthesis/patts
/dev/loop6
10240    10240                0 100%
/usr/share/chromeos-assets/quickoffice/_platform_specific

Thanks!

closed 2021-07-01 by darethehair