sebanc/brunch #1097: ROOT-A was not cleanly unmounted

dimasirfandanu, 2021-06-07:
Hello,

I have problem with my chrome os installation is stuck at chrome logo and going on boot loop 😔. From boot log I always see there is an error of ROOT-A partition "ROOT-A was not cleanly unmounted". Boot script automatically fixed this error with fsck but still can't pass the chrome os logo. I've done fsck with live Linux but still everytime it's going into chrome os boot process there is always something wrong with ROOT-A partition. 

Your help is greatly appreciated, Thank you!

smartguy1196, 2021-06-12:
Try doing this, so they can get more details: https://github.com/sebanc/brunch/issues/756#issuecomment-752710319

They need more information, before they can figure out what went wrong

dimasirfandanu, 2021-10-01:
as https://github.com/sebanc/brunch/issues/331#issue-642332549 need to be careful before logging to brunch even tough I use a tablet with emmc sometimes brunch installation will break if I login to fast into Chrome OS. But sometime running fsck can fix the problem.