sebanc/brunch #1102: Can not Sleep when I close the laptop。HP Envy13 x360。

wahahaha1314, 2021-06-08:
Brunch, 90.13816.0

Chrome OS  90.0.4430.218（release） （64bit）

zork_recovery.

CPU：AMD Ryzen 7 4700U .
Ram: 16GB

grub-config ：
menuentry "ChromeOS" --class "brunch" {
        linux ($disk,7)/kernel-5.10 boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug options=acpi_power_button \
                console= vt.global_cursor_default=0 brunch_bootsplash=default quiet
        initrd ($disk,7)/lib/firmware/amd-ucode.img ($disk,7)/lib/firmware/intel-ucode.img ($disk,7)/initramfs.img
}

When I close the laptop, the fan is still working .and It 's lost 80% battery power over a night.
Even I close the WIFI and Bluetooth.

I allready try 'suspend_s3' ,but it's not work.

Can someone give me an idea 。。

mrwhistles, 2021-07-29:
This is a problem across the board for Renoir based AMD cpus where s2idle is not properly implemented for them. Kernel 5.14 is looking to include all relevant patches by default. 

See https://gitlab.freedesktop.org/drm/amd/-/issues/1230

We would either need the series of patches applied to the 5.10 kernel or hopefully when 5.14 is officially released we can have that included in brunch.

Weddin, 2021-11-29:
I can confirm that sleep is not working on Chuwi hi 10 pro (z8350) with kernel 5.10. On 5.4 sleep is working