sebanc/brunch #1108: Touch screen issue

czyun44, 2021-06-11:
Hi 
I have a Jumper Ezpad 6 pro, not the b version. 
Have a problem with the touchscreen.
Managed to find the correct driver for my tablet ([https://github.com/onitake/gsl-firmware](https://github.com/onitake/gsl-firmware))
Then I went on to load the firmware into the /lib/firmware/silead. 
At the first few tries, it fails, but after I rename the firmware file from 'gsl3692-jumper-ezpad-6-pro.fw' to 'gsl3692-jumper-ezpad-6-pro**-b**.fw'

The firmware loads and it gets inverted y-axis touch instead. 

As I check the source file of the touchscreen parameter (at 'kernel-5.4 or 5.10\drivers\platform\x86\touchscreen_dmi.c'), 
the difference in both versions that identify the device is the bios-date. 
[image]

The reason why it gets an inverted y-axis is that the B version is set with an inverted y-axis parameter 
[image]

I also check my bios date for my tablet and it has the non-'b' version (bios date is 08/18/2017)
Retreived from the command DMIDECODE in ChromeOS.
[image]

I am not sure why it has this issue(being recognized as B version), but maybe the DMI_MATCH doesn't work (use DMI_EXACT_MATCH instead?) Or a date localization issue?
Or could use Firmware Revision as another match variable?

I have tried to rebuild the brunch framework (with some fixes I have stated above) myself from the source code (this guide '[https://github.com/sebanc/brunch/blob/master/BUILDING.md](https://github.com/sebanc/brunch/blob/master/BUILDING.md)')  and failed at producing the release package. 

Could you help to fix it in the next release?

sebanc, 2021-06-12:
I added a few parameters in brunch kernels to calibrate silead touchscreens:

```
+static int custom = 0;
+module_param(custom, int, 0644);
+MODULE_PARM_DESC(custom, "Override touchscreen settings");
+
+static int x = 0;
+module_param(x, int, 0644);
+MODULE_PARM_DESC(x, "Override touchscreen x size");
+
+static int y = 0;
+module_param(y, int, 0644);
+MODULE_PARM_DESC(y, "Override touchscreen y size");
+
+static int min_x = 0;
+module_param(min_x, int, 0644);
+MODULE_PARM_DESC(min_x, "Override touchscreen x minimum");
+
+static int min_y = 0;
+module_param(min_y, int, 0644);
+MODULE_PARM_DESC(min_y, "Override touchscreen y minimum");
+
+static int fuzz_x = 0;
+module_param(fuzz_x, int, 0644);
+MODULE_PARM_DESC(fuzz_x, "Override touchscreen x fuzz");
+
+static int fuzz_y = 0;
+module_param(fuzz_y, int, 0644);
+MODULE_PARM_DESC(fuzz_y, "Override touchscreen y fuzz");
+
+static int max_pressure = 0;
+module_param(max_pressure, int, 0644);
+MODULE_PARM_DESC(max_pressure, "Override touchscreen max pressure");
+
+static int fuzz_pressure = 0;
+module_param(fuzz_pressure, int, 0644);
+MODULE_PARM_DESC(fuzz_pressure, "Override touchscreen fuzz pressure");
+
+static int invert_x = 0;
+module_param(invert_x, int, 0644);
+MODULE_PARM_DESC(invert_x, "Override touchscreen x inversion");
+
+static int invert_y = 0;
+module_param(invert_y, int, 0644);
+MODULE_PARM_DESC(invert_y, "Override touchscreen y inversion");
+
+static int swap_x_y = 0;
+module_param(swap_x_y, int, 0644);
+MODULE_PARM_DESC(swap_x_y, "Override touchscreen x/y swap");
```

Pass "of.cutom=1" on the kernel command line along with the parameters you want to override, in your case probably "of.custom=1 of.invert_y=1"

czyun44, 2021-06-12:
Still not working after adding the "of.custom=1 of.invert_y=1"  or  "of.invert_y=0" after "cros_debug" and before "options=". 
Issue still same, Inverted y-axis touchscreen

sebanc, 2021-06-12:
My mistake, could you try "of_touchscreen.custom=1 of_touchscreen.invert_y=1"

czyun44, 2021-06-13:
Yes, It works after placing "of_touchscreen.custom=1 of_touchscreen.invert_y=1" after "cros_debug" and before "options=".

Thank you very much!@sebanc

closed 2021-06-13 by czyun44