sebanc/brunch #1109: not enough space to create image file

Moon-Haze, 2021-06-11:
When I install chrome os by using Ubuntu 20.04, I happened an error which is "not enough space to create image file".

simplyrajatgupta, 2021-06-12:
@Moon-Haze      Restart your system

princepjamie06832, 2021-06-12:
How much space you'll need to create image file on your drive or partition?

simplyrajatgupta, 2021-06-13:
> How much space you'll need to create image file on your drive or partition?

14 gb for img and 1 gb for file system and other system information
# **(=15 GB)**

Moon-Haze, 2021-06-13:
The resered space is 49G.

Moon-Haze, 2021-06-13:
[image]
I think this can try.

Moon-Haze, 2021-06-13:
Thank you very much.

simplyrajatgupta, 2021-06-13:
Please close your issue

simplyrajatgupta, 2021-06-13:
@Moon-Haze  Please close your issue

closed 2021-06-13 by Moon-Haze