sebanc/brunch #1112: Low-Power Charger Connected Issue

bpvarsity, 2021-06-15:
I am using the oem charger and a 3rd party and both cause ChromeOS to say a lower power charger is connected.
Is this a bug and is there anyway to disable the low-power charger connected notification?

Thanks

kushmittal2009, 2021-06-18:
You can by clicking the x next to the notification about the low power and it will be gone.

skiv71, 2021-07-11:
I've seen this also, but only when running kernel-5.10

chas0001, 2021-09-23:
I have the exact same problem with ChromeOS 92, Volteer and Kernel-5.10.    Windows and Ubuntu recognise the charger as quick charge.

skiv71, 2021-11-01:
I've just a request to make 5.10 the default kernel... surely not a good idea whilst this issue exists?

I'm currently running 5.10 as the latest r94 chrome os (via PWA) borked my webcam (so I can confirm this is still an issue)

closed 2026-02-24 by github-actions[bot]