sebanc/brunch #1114: [DELL Venue 11 Pro 7140] No audio

qtotter, 2021-06-17:
I guess kernel don't support my PC?  

****
DELL Venue 11 Pro 7140 (tablet): 
CPU: Intel core M-5Y10 (Broadwell)
Audio: Realtek HW Audio Codec/Intel Smart Sound Technology (from Device Manager of Win10)

Brunch: brunch_r90_stable_20210523.tar.gz
ROM: wizpig 
Kernel: 5.4
dmesg.txt: [https://drive.google.com/file/d/1IjUx-xGPXrOpruFJE356_wrLJMPbPOuc/view?usp=sharing](url)
****

Almost everything works fine, including touchscreen, WiFi, BT, rotation, back camera, etc., except sound (and front camera, which is not really important for me.) What can I do with it?

Thanks to sebanc's continuous development, my Dell Inspiron 14 2-in1 (5400) now runs Chrome OS flawlessly with kernel 5.10! So, I bought a second-handed Venue 7140 Pro to use as a tablet, but I am facing this issue now. Any supports will be appreciated. Thanks!

qt

sebanc, 2021-06-19:
Try kernel 5.10 first and if it does not work try kernel 5.10 with "disable_intel_hda" option.

qtotter, 2021-06-19:
Hi sebanc, I tried 5.10 and 5.10+disable_intel_hda, and the result was the same, still no sound...

qtotter, 2021-06-19:
```
$ grep -i codec /proc/asound/card*/codec#*
Codec: Intel Broadwell HDMI
```
I guess Intel Broadwell HDMI is for HDMI, but Realtek HW Audio for speaker is not recognized as a codec chip...

qtotter, 2021-06-19:
broadwellrt286 is recognized as a card on brunch kernel, but it doesn't seem to be working. Audio is working fine on ubuntu...

qtotter, 2021-06-20:
Reading [here](https://bbs.archlinux.org/viewtopic.php?id=208461&p=2), the command line, 
`$ amixer -c 1 set SPO on`
also works with Chrome OS, and Venue 11 Pro 7140 now plays sound. I hope this can be fixed at the kernel level...

sebanc, 2021-06-20:
I think the issue is in "alsa-lib", chromeos currently uses a version which is quite old and is known to have some issues so I cannot really do much about it.

However, that command is a nice find and you can easily create a bootscritpt to enable it on startup by running the below commands in crosh shell:
```
sudo mkdir /var/brunch/bootscripts
echo "amixer -c 1 set SPO on" | sudo tee /var/brunch/bootscripts/sound.sh
sudo chmod 0755 /var/brunch/bootscripts/sound.sh"
```

Bootscripts persists across updates so you should only need to run those commands once.

qtotter, 2021-06-20:
Hi sebanc, Thanks for the boot script tip!

qt

closed 2021-06-20 by qtotter