sebanc/brunch #1115: is chrome os supported on nvidia?

kingmaj0r, 2021-06-18:
Title

princepjamie06832, 2021-06-18:
There's no NVIDIA support for Chrome OS. Only supported for Intel iGPU and AMD APU.

UmashankarTriforce, 2021-06-19:
Well it's complicated. You can install chrome os on a device if you got an NVIDIA dGPU. But don't expect the display to work from the dGPU as chrome os does not have the drivers for NVIDIA dGPU. If it's like a laptop with an iGPU and an NVIDIA dGPU, you can install chrome os, and disable the dGPU with some kernel parameters. You can also use the NVIDIA dGPU on CUDA compute mode, by using [Brioche](https://github.com/sebanc/brioche) and install the drivers. Since there's no display server attached to the dGPU, don't expect any 3D applications to run on dGPU.

lucasgabrielmoreno, 2021-07-26:
Yes, Geforce Now:
https://www.nvidia.com/es-ar/geforce-now/

It's still in beta for Chrome OS

trusktr, 2022-09-05:
> Yes, Geforce Now:

I believe this thread is about installing Chrome OS on a device that has an Nvidia GPU.

Geforce Now is **_not_** Chrome OS running on a computer with an Nvidia CPU. [Geforce Now](https://geforcenow.com) is a service owned by Nvidia that streams application graphics over the network to your system, exactly like [Google Stadia](https://stadia.google.com/). This is why you can run Geforce Now on _**any**_ device simply by [loading Geforce Now inside your browser](https://play.geforcenow.com/) in your browser tab on any device, which is a lot different than Chrome OS on a laptop with Nvidia GPU.

closed 2022-09-11 by kingmaj0r