sebanc/brunch #1116: Custom Kernel Support

UmashankarTriforce, 2021-06-19:
This is just an experiment. I'm aware the current version of brunch ships with kernel-4.19, kernel-5.4, and kernel-5.10. I was just wondering if I can pick up any upstream kernel, compile it, and use it on brunch? I know some config has to applied while compiling the kernel to make it compatible with ChromeOS w/Brunch. This gives the opportunity to fine tune the kernel for a specific device or even compile the kernel with extra flags and features. 

I just want to know what steps are necessary to get this working.

sebanc, 2021-06-19:
As far as I can tell, the main difference between the chromiumos kernel (https://chromium.googlesource.com/chromiumos/third_party/kernel/+refs) and the mainline kernel is specific patches for android support.

As there are only chromiumos kernel sources for LTS releases, to use a non-LTS mainline kernel version you would have to identify all necessary android patches and port them over.

As per your other issue, ARCVM and crostini are VMs made by Google with their own kernels (the kernel version does not need to match and is not as important anyway as it is just to run a virtual machine). You can probably update the vm kernel if you want but that's not the purpose of this project (i believe the ARCVM and crostini kernel configs are also in the chromiumos kernel repository)

UmashankarTriforce, 2021-06-20:
Thanks for your input! I'll try to work on it!

closed 2021-06-20 by UmashankarTriforce