sebanc/brunch #1118: How can I remove Chrome and reinstall Windows?

neo338388, 2021-06-19:
Laptop: SP4
Installed the Brunch ChromeOS for many months, after encounterd many issues I would like to reinstall Window, could you tell me how to do it?  Thanks.

kerta1n, 2021-06-21:
Grab a USB stick, visit [this site](https://www.microsoft.com/en-us/software-download/windows10ISO), and then install the [Recovery extension](https://chrome.google.com/webstore/detail/chromebook-recovery-utili/jndclpdbaamdhonoechobihbbiimdgai/RK%3D2/RS%3DUI2uA8SxDAwF_T9oPb4YviZFT3Y-). Go to files, rename the ISO to a  IMG file, and launch the Recovery extension. Hit the settings cog in the top right, and click "use local image", then select that Windows.img file you just renamed. Let it flash, and reboot into the thumbdrive, which should lead into the Windows installer.

neo338388, 2021-06-21:
There are 12 partition in the HardDrive now, should I delete them all?  As window setup said can't be installed on these partitions.

kerta1n, 2021-06-30:
Good to hear!

Buzzcut22, 2021-07-09:
do you have the step by step procedure for this?

please send por post here.

kerta1n, 2021-07-11:
> do you have the step by step procedure for this?
> 
> please send por post here.

This is already posted. Please read my first reply.

closed 2026-02-24 by github-actions[bot]