sebanc/brunch #1124: Linux (beta)

FilipeF85, 2021-06-23:
Hi i have installed "Brunch r91 stable 20210620" with the rammus recovery "chromeos_13904.55.0_rammus_recovery_stable-channel_mp-v2.bin.zip"
I have install it directly to single disk.
I have play store working everything fine, but i cannot activate the Linux (beta) option on Developers i turn it on then show the windows informing that will install i do next i select the username the space i let the recommended that is 7.5GB, press Install then i have this error:

Error installing Linux...

Error starting the virtual machine. Please try again.

forumi0721, 2021-06-24:
Me too.
Any others ok?

skiv71, 2021-06-24:
Me and well... just tried cloudready, which was r89... worked ok there.

iknoors, 2021-06-24:
hey i just saw this thread, is there linux for family link yet

rocketprogrammer, 2021-06-26:
Happening for me too.

toledompm, 2021-06-26:
Same thing here (intel celeron n4000), tried some brunch release - chrome os recovery images combos. With this one I was able to install the linux beta:
- Brunch: r91 stable
- Rammus: 90
I'm kinda of a noob when it comes to brunch though, are there any drawbacks to using an older recovery image or will I be fine?

BiteDasher, 2021-06-27:
Me too

Recovery: rammus 91
Brunch: r91

mm-a, 2021-06-28:
me too -  r91

bhavishyamadan, 2021-06-29:
Set this flag to “disabled” 
By typing this in chrome

chrome://flags#crostini-use-dlc

Now 
Reboot and try again

mm-a, 2021-06-29:
> Set this flag to “disabled”
> By typing this in chrome
> 
> chrome://flags#crostini-use-dlc
> 
> Now
> Reboot and try again

that worked for me, thank you

FilipeF85, 2021-06-29:
> Set this flag to “disabled”
> By typing this in chrome
> 
> chrome://flags#crostini-use-dlc
> 
> Now
> Reboot and try again

Just tested your solution and it worked fine.
Thanks for your help.

closed 2021-06-29 by FilipeF85

compnew2006, 2021-07-10:
> Set this flag to “disabled”
> By typing this in chrome
> 
> chrome://flags#crostini-use-dlc
> 
> Now
> Reboot and try again

It's Work as chrome  thanks for help.