sebanc/brunch #1125: Random touches on touch screen

midozalouk, 2021-06-23:
Hello
i sometimes has random touches on my touch screen !!!
i am using hatch R91 recovery on surface pro 6
with the latest brunch Brunch r90 20210523
how can i solve that or investigate it ?
I use the this grub configuration

menuentry "ChromeOS" { img_part=/dev/nvme0n1p5 img_path=/ChromeOS.img search --no-floppy --set=root --file $img_path loopback loop $img_path linux (loop,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \ cros_secure cros_debug enforce_hyperthreading=1 options=enable_updates,pwa,ipts,suspend_s3,no_camera_config,mount_internal_drives loop.max_part=16 img_part=$img_part img_path=$img_path \ console= vt.global_cursor_default=0 brunch_bootsplash=default quiet initrd (loop,7)/lib/firmware/amd-ucode.img (loop,7)/lib/firmware/intel-ucode.img (loop,7)/initramfs.img }

ZnlWomble, 2021-07-02:
I'm having the same issue on the Surface Pro 5 with brunch r91

midozalouk, 2021-07-02:
> I'm having the same issue on the Surface Pro 5 with brunch r91

I noticed it happens after i close the lid and reopen it!!! 
I should restart to make it work

sebanc, 2021-07-02:
For Surface Pro 4, 5, 6 and Surface Book 1 & 2, I would recommend using the 4.19 kernel (see https://github.com/sebanc/brunch/issues/689#issuecomment-739703874)

closed 2026-02-24 by github-actions[bot]