sebanc/brunch #1126: Wifi bcm94352hmb not working after update to cros 91 and brunch 91.

b1t0y, 2021-06-24:
Hi. Is there any solution to get my wifi back working. No issues with brunch and chrome os 90 version, but stopped working and cannot find any network. I can turn the wifi on and off but cannot find any network. Wifi icon wifi with x on the inside.  Tried putting broadcom_wl and without same issues. Please help! Thanks in advance.

TrickLin, 2021-07-10:
this might help
https://www.mail-archive.com/packman@links2linux.de/msg11832.html

lucasgabrielmoreno, 2021-07-21:
Same problem, bluetooth working but not wifi

ravindrank, 2021-08-09:
Even I have the same issue.. I have added the options and it was working fine until few months ago. Over last few months, the wifi doesnt work any more. I'm sure lot of Broadcomm  users have the same. The module loads fine and I can see the drivers loaded but no wifi accesspoints show up post scanning

b1t0y, 2021-08-12:
> Even I have the same issue.. I have added the options and it was working fine until few months ago. Over last few months, the wifi doesnt work any more. I'm sure lot of Broadcomm users have the [...]

So did it ever work for you since you updated chrome os 91 or 92?  Coz i tried brunch 91 and chrome os 90 and wifi was working. Its when it chrome os 91 or 92. So i don't know if it is a problem with the new chrome os or linux broadcom wifi driver?

ravindrank, 2021-08-18:
It never worked probably since 90...not sure as I lost track of it.. didnt troubleshoot much .. I even tried an older build of brunch after seeeing some suggestions....so in all probably it is due to the new kernel changes...

b1t0y, 2021-10-03:
WIFI working again!  Just an update for everyone having this issue with this wifi card.  Since updating to latest brunch 93 and chrome os 93, wifi is working again.  Thank you so much!!!  Happy to have a working chrome os.