sebanc/brunch #1131: Error installing Linux: Error installing Linux

siumingdev, 2021-06-26:
Error message with title "Error installing Linux" and content "Error downloading the virtual machine. Please try again." shows up during initial setup of Linux (beta).

I am pretty sure there is no problem on the internet access.

[image]

siumingdev, 2021-06-26:
Brunch: r91 stable 20210620
ChromeOS image: rammus 91.0.4472.114
Hardware: I guess not related?

siumingdev, 2021-06-26:
per suggestion from sebanc in Telegram, downgraded to r90 and it works now.

closed 2021-06-26 by siumingdev

lucasgabrielmoreno, 2021-07-26:
> per suggestion from sebanc in Telegram, downgraded to r90 and it works now.

Same problem from r91! Works downgrade to r90
Thank you!!!