sebanc/brunch #1137: [DELL Venue 11 Pro 7140] Temporary freeze on boot

qtotter, 2021-07-02:
****
DELL Venue 11 Pro 7140 (tablet)
CPU: Intel core M-5Y71 (Broadwell)

Brunch: brunch_r91_stable_20210620.tar.gz
ROM: cave
Kernel: 5.10
****

Hi sebanc, 

Chrome OS works pretty well on DELL Venue 11 Pro 7140 except internal microphone (not a big issue for me, and I know it's not easy to fix this), high battery consumption while sleep (I don't think it actually sleeps), and temporary freeze on boot. 

My focus here is the last one. Every time you boot Chrome OS, it freezes for 20-30 seconds on login screen. After 20-30 second waiting, you can use mouse/touchpad/touchscreen/keyboard and type passwords to login. This happens only on boot ONCE, not when resuming from sleep. 

Precise speaking, it's more like inputs stop working after a couple of seconds on login screen and restart working 20-30 seconds. Any ideas about this small temporary freeze on boot? Thanks in advance. 

qt

sumcda, 2021-11-06:
Did you ever find out what the issue was with the freezing at login screen? I get the same issue with my Thinkpad Yoga 12.

qtotter, 2021-11-07:
Unfortunately, I haven't found a solution, and moved back to Windows 10 (and 11).

I thought it could be a better solution to run Android apps on Windows 11 because everything works fine on Windows 11 and Android apps now run on it (with Play Store.) The only issue with running Android apps on Windows 11 is it needs more RAMs...

sumcda, 2021-11-07:
I moved back to Windows 11 also. Mainly to try out the android apps but I couldn't get it to work smoothly. I also have 8gb ram. I moved back to brunch as the frozen login issue was really the only problem I had and I can live with it as I mostly boot from sleep mode.

When I went back to Windows 11 I installed brunch on a bootbale usb and the issue with the frozen login screen did not occur.   There were other issues with the usb brunch which is why I ended up not pursuing that method but having seen the login working on my laptop I believe there must be a fix for this issue. If I find one I will let you know here.

sebanc, 2021-11-07:
Could you post a dmesg ? (run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and attach dmesg.txt from your Downloads folder)

sumcda, 2021-11-07:
Here it what you requested sebanc. Appreciate it. 

[dmesg.txt](https://github.com/sebanc/brunch/files/7492510/dmesg.txt)

sebanc, 2021-11-07:
Could you try to boot with "module_blacklist=intel_vbtn" added on the kernel command line ?

sumcda, 2021-11-07:
I can certainly try that if you are prepared to give me some pointers on how to do it.

sebanc, 2021-11-07:
Press "e" when you see grub and add a space and the above text after "cros_debug"

sebanc, 2021-11-07:
Then try with "link_power_management_policy=min_power"

sumcda, 2021-11-07:
I don't see grub on boot. I have mentioned this before is Discord or Telegram but because I found the brunch-toolkit I did not follow it up. Is there another way?

sebanc, 2021-11-07:
From chromeos, run "sudo edit-grub-config" and it will ask you at some point if you want to add custom kernel command line parameters. Enter the above parameters to test there.

At a later stage, could you try this build and check if you see the grub menu with it ?
https://github.com/sebanc/brunch-unstable/releases/download/r94-unstable-20211107/brunch_r94_unstable_20211107.tar.gz

sumcda, 2021-11-07:
I am currently using that unstable brunch. I will restart the laptop to see if I can see grub.

sumcda, 2021-11-07:
There is no grub on boot. It goes straight to Loading brunch framework screen. When I run "sudo edit-grub-config" I get the following screen:

[image]

sebanc, 2021-11-07:
Ok, this is the configuration menu, on the 5th page it will ask you for custom kernel parameters

sumcda, 2021-11-07:
It says page 1/2 so I didn't realise there were more pages. 

When I get to the kernel parameters page I paste the commands you gave me above. After each one the pc restarted. Is this correct? If so there was no change to the issue. The pc started normally with the loading brunch framework screen. There was no rebuilding framework screen that I would get if I was adding a config like enable_updates etc.

sebanc, 2021-11-07:
if you still have "link_power_management_policy=min_power", could you post another dmesg ?

sumcda, 2021-11-07:
link_power_management_policy=min_power was the last one I tried but I have no idea if it even added it or how I would remove it.

[dmesg.txt](https://github.com/sebanc/brunch/files/7492608/dmesg.txt)

sebanc, 2021-11-07:
Try this "ahci.mobile_lpm_policy=0"

sebanc, 2021-11-07:
Between, I can confirm what you did previously was correct. Just re-run the setup menu (it resets the config) and try with the above one.

sumcda, 2021-11-07:
No change. I get about 5 blinks of the cursor. Then the cursor disappears for about 20 seconds. Then the cursor starts blinking again and I can enter the password. 

[dmesg.txt](https://github.com/sebanc/brunch/files/7492640/dmesg.txt)

FYI there is also a smaller message to the left that says something about bluetooth being switched off. I don't think I have any issues with bluetooth otherwise so this may have nothing to do with it.

sebanc, 2021-11-07:
Maybe try "module_blacklist=thinkpad_acpi"

sumcda, 2021-11-07:
I know nothing about these things other than when I booted brunch from a bootable usb, after installing Windows 11, this issue was not present. I have since clean installed brunch on the pc. Is there something in the bios I can try?

I will try the module_blacklist=thinkpad_acpi now.

sumcda, 2021-11-07:
No change

qtotter, 2021-11-07:
Is it just a coincidence that sumcda's PC and mine have both intel 5th gen CPUs?

sebanc, 2021-11-07:
I think it is related to this:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/550559

Try "libata.force=1:noncq"

sumcda, 2021-11-07:
Same outcome

[dmesg.txt](https://github.com/sebanc/brunch/files/7492664/dmesg.txt)

Apologies to qtotter if I hijacked your thread. Hopefully you are still interested even though you moved back to Windows.

qtotter, 2021-11-07:
sumcda, no prob. I might try chromeos again if things work well.

sebanc, 2021-11-07:
Maybe "libata.force=3.0G"

Source: https://bugzilla.kernel.org/show_bug.cgi?id=49841

You might also check if you have something called "SMART mode" in your bios and disable it if it is the case.

sumcda, 2021-11-07:
Still no luck with this. I checked bios but there was no SMART mode as far as i could tell. Unless anything else springs to mind immediately please feel free to get on with other things and we can have another go at this when you have more time.

sebanc, 2021-11-07:
One last for today: "libata.force:noncq"

My only guess is that it is related to libata, everything else seems generally correct in the dmesg as far as I can tell.

sumcda, 2021-11-07:
That did not work unfortunately but I appreciate your continued help sebanc. If you think of anything else in the mean time post it here and I will test it out.

sumcda, 2021-11-08:
FYI. I decided to update the kernel to 5.10. It had no effect on the boot issue but when I enter "sudo edit-grub-config" in shell it outputs "killed". 

I would return to 5.4 but I can't get to the configuration menu now.

migsyboy, 2022-01-21:
I confirm I have the same issue at login with a DELL Venue 11 Pro 7140

sumcda, 2022-01-21:
Unfortunately I dropped a full cup of tea over my ThinkPad. On the plus side I don't have to try and work out a fix to this issue anymore lol. When I get a replacement I will be installing brunch again so hopefully there will be no issues. If you want to try anything yourself i'm sure sebanc will be happy to help out. All the best to you. Feel free to ask me anything as I tried alot of different things.

migsyboy, 2022-01-22:
Headphones don't seem to work either.

migsyboy, 2022-02-11:
Looks like the microphone doesn't work either.

@sebanc your product is amazing, thanks. Do you think the new release would solve these issues?

migsyboy, 2022-02-11:
I have just updated to the latest release, and issues persist unfortunately

moscard, 2022-03-15:
Same issue here on a ThinkPad X1 Carbon 4th gen
CPU i5 6200U
Ramus : Brunch r97, ChromeOS r99
After the freeze at login, it works like a charm.

moscard, 2022-03-16:
ok some more info:
Same PC ThinkPad X1
- Tried Flex OS -> hang at boot after freeze
- Tried Cloudready -> same behaviour, freeze at logging for a while and then work flawlessly
- When I boot from USB key on same PC -> no issues.

So, I suppose it has something to do with hard disk.

moscard, 2022-03-16:
The SSD is a LiteON L8H-256V2G
I tried to update the firmware under windows with Lenovo tool, it says that it is up-to-date. I found no way to update it under Linux.

moscard, 2022-03-19:
It seems it is an old issue related to LITEON SSDs
https://issuetracker.google.com/issues/172211901?pli=1
It should be fixed from a while, but no.

note : I found a way to update it under Linux, but there is no update available for my LiteOn disc

moscard, 2022-03-26:
Ok, -> SOLVED
I exchanged the SSD to an "innovation IT" one, no more issues.

migsyboy, 2022-03-26:
Hi!

Would you mind explaining what you did exactly?

Thanks!

moscard, 2022-03-26:
Simple, as explained, I had a LiteON SSD which was the cause of the issue.
I purchased a new SSD "Innovation IT" and changed it. It could be another brand, the problem is the LiteON one as explained here: https://issuetracker.google.com/issues/172211901?pli=1

migsyboy, 2022-03-26:
Ok thanks. I am not sure I can actually open my Dell laptop and change the
SSD drive (and even if I could, I don't think I'd want to spend more money
on it) so I'll just have to put up with this issue. Thanks!

moscard, 2022-03-26:
Sure you can, and the SSD is about 30$, so probably it worth it.
Anyway, is a known issue by LiteON and for some kinds they have a firmware upgrade, you could try that before.
On mine, LiteON L8H-256V2G, there were no available updates, so I preferred to change it. 
To be clear, the issue appears only with all version of Chrome OS, not with Windows nor Linux.