sebanc/brunch #1147: Volteer - Google's recovery images are currently down.

riqueps, 2021-07-13:
Hello there, I'm trying to download volteer recovery images but it's showing me the following message: "Google's recovery images are currently down. They will return here as soon as they are up again. Sorry for the inconvenience!"
Someone could tell me when it will return or if someone has a link to let me download it ?
Thanks a lot.

WesBosch, 2021-07-13:
Currently, this link still works
https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_13904.55.0_volteer_recovery_stable-channel_mp-v6.bin.zip

riqueps, 2021-07-13:
> Currently, this link still works
> https://dl.google.com/dl/edgedl/chromeos/recovery/chromeos_13904.55.0_volteer_recovery_stable-channel_mp-v6.bin.zip

Thanks a lot!!

closed 2021-07-13 by riqueps