sebanc/brunch #1148: Dual booting with Pop_OS (uses systemd-boot)

umasse, 2021-07-14:
Hello,

I have a laptop installed with Pop_OS, which uses systemd-boot by default. I want to have it dual-boot, with ChromeOS+brunch installed in an image file. The process worked fine, and I have ChromeOS installed into an image file in the main partition.
However, the instructions presented are only for GRUB.

I see 3 possible options:
1. Figure out how to boot ChromeOS from inside the image using systemd-boot.
2. Install Grub in Pop_OS, but this will break Pop_OS in several ways, and is officially discouraged and unsupported.
3. Install Grub in Pop_OS, but still boot using systemd-boot. Chainload Grub from systemd-boot, to be able to boot ChromeOS.

Does anyone have any experience with Pop_OS and Brunch?
Thanks!

sebanc, 2021-07-14:
In general, it is easier to use the linux distro grub to dual boot but in that case you will need to create a separate bootloader (some specific grub features are needed for dual boot).

1) You would need to mount the brunch efi partition (partition 12) and copy the /efi/boot folder to /efi/brunch in your hdd efi partition.
2) replace "grub.cfg" content with the grub config provided by the chromeos-install.sh script
3) Use "efibootmgr" from linux to create an uefi boot entry named "brunch" pointing to /efi/brunch/grubx64.efi
4) (optional) you can probably create an additional boot entry to boot Pop_OS systemd-boot efi file from grub

umasse, 2021-07-15:
Thanks! This seems doable. I will check if I can just install grub in Pop_OS to have the files without replacing systemd-boot. That may be easier.