sebanc/brunch #1155: Mic not working please help

gautamsharma1512, 2021-07-16:
I hace installed chrome os on my device but internal mic not working please help. please its urgent. please.

ameyjoshi3068, 2021-07-17:
Gautam ji,
I am just a normal user of brunch, but I can suggest you to add internal_mic_fix option.

https://github.com/sebanc/brunch#framework-options

I believe it will fix your issue.

Twilighttony, 2021-07-18:
> Gautam ji,
> I am just a normal user of brunch, but I can suggest you to add internal_mic_fix option.
> 
> https://github.com/sebanc/brunch#framework-options
> 
> I believe it will fix your issue.

That didn't resolve my issue either when it comes to internal mic. I also noticed that even when using a headset with mic attached to my laptop, I cannot retrain the google assistant either.