sebanc/brunch #1159: Broadcom BCM94352HMB Wifi not working

lucasgabrielmoreno, 2021-07-21:
I've Broadcom BMC 9 4352 HMB.
Bluetooth works! But can't make work wifi.

Methods I've tried:
1. broadcom_wl at grub _(with this video: https://www.youtube.com/watch?v=HBlfUnLKFNA)_
2. Install Broadcom Linux driver in Debian 10 Linux beta _(with this video: https://www.youtube.com/watch?v=JW0f3NOjWys)_
3. Try to inject wl.ko at kernel like in old Chromium OS _(with this old method: https://arnoldthebat.co.uk/wordpress/2013/02/24/how-to-get-the-broadcom-wireless-drivers-working/comment-page-1/)_
4. Enable read to root at /lib/firmware/b43 folder into ROOT-C partition in Linux Mint (my no sense method)

Thanks! brunch is awesome!

lucasgabrielmoreno, 2021-07-26:
Still trying in Linux deb (Debian 10) console:

sudo su
sudo apt-get update
sudo apt-get upgrade
sudo apt-get purge firmware-b43-installer   
sudo apt-get install dkms
sudo apt-get install libc6-dev
sudo apt-get install linux-libc-dev
sudo apt-get install software-properties-common
sudo add-apt-repository "deb http://archive.ubuntu.com/ubuntu $(lsb_release -sc) main universe restricted multiverse"
sudo apt-get update
mkdir bcmwl && cd bcmwl
wget http://mirrors.kernel.org/ubuntu/pool/restricted/b/bcmwl/bcmwl-kernel-source_6.30.223.271+bdcom-0ubuntu7_amd64.deb
sudo dpkg -i bcmwl*.deb

lucasgabrielmoreno, 2021-07-26:
In Chrome Os console:

sudo nano /etc/modprobe.d/broadcom_wl.conf

Add this lines:
blacklist bcm43xx
blacklist brcmfmac

lucasgabrielmoreno, 2021-07-26:
FIXED!
Solution: downgrade until newer fixed versions.

Downgrade to r90 (both, Chrome OS and brunch)
Brunch and recovery versions must be the same.

- Into grub add: 
options=broadcom_wl

- reboot

Thanks!

closed 2021-07-26 by lucasgabrielmoreno

b1t0y, 2021-08-08:
> FIXED!
> Solution: downgrade until newer fixed versions.
> 
> Downgrade to r90 (both, Chrome OS and brunch)
> Brunch and recovery versions must be the same.
> 
> * Into grub add:
>   options=broadcom_wl
> * reboot
> 
> Thanks!

Did you find any solution for the wifi on chrome os 92? Thanks

lucasgabrielmoreno, 2021-08-08:
> > FIXED!
> > Solution: downgrade until newer fixed versions.
> > Downgrade to r90 (both, Chrome OS and brunch)
> > Brunch and recovery versions must be the same.
> > 
> > * Into grub add:
> >   options=broadcom_wl
> > * reboot
> > 
> > Thanks!
> 
> Did you find any solution for the wifi on chrome os 92? Thanks

Nope! still in 90

lucasgabmoreno, 2022-03-15:
> I found a way to solve this "problem":
> 
> 1. Enter Brunch "Chrome OS(Settings)"
> 2. Select broadcom_wl option
> 3. The system will reboot then rebuild Chrome OS RootFS
> 
> BCM94352HMB works :)

Yes, it's fixed in last brunch versions!