sebanc/brunch #1162: Error installing Linux Error downloading the virtual machine. Please try again.

harshcans, 2021-07-23:
" Error installing Linux
Error downloading the virtual machine. Please try again. "

This error is showing in my screen.

Rammus 91

bm-a, 2021-07-23:
> " Error installing Linux
> Error downloading the virtual machine. Please try again. "
> 
> This error is showing in my screen.
> 
> Brunch: r91 stable 20210620
> ChromeOS image: rammus 91.0.4472.114

Set this flag to “disabled”
By typing this in chrome

chrome://flags#crostini-use-dlc

Now
Reboot and try again

harshcans, 2021-07-24:
Thank u @bm-a for this awesome reply .
" It's Work "

I have been trying tying theese from so many days.

closed 2021-07-24 by harshcans

block6791, 2021-08-01:
> > " Error installing Linux
> > Error downloading the virtual machine. Please try again. "
> > This error is showing in my screen.
> > Brunch: r91 stable 20210620
> > ChromeOS image: rammus 91.0.4472.114
> 
> Set this flag to “disabled”
> By typing this in chrome
> 
> chrome://flags#crostini-use-dlc
> 
> Now
> Reboot and try again

Solved by disalbing this flag. Thanks for the helpful tip.

yeahman45, 2022-06-04:
I am having the same error with [Brunch r101 stable 20220529]

chrome://flags#crostini-use-dlc
does not seem to solve the issue.

King-Lucifer6, 2022-06-18:
Error installing Linux
Error downloading the virtual machine. Please try again.

ChromeOs Version 101.0.4951.59 (Official Build) (64-bit)
[image]
and theres no #crostini-use-dlc in my ChromeOs.

Can anyone please help?

ankur-connect, 2022-07-11:
Having the same issue @King-Lucifer6

block6791, 2022-07-12:
I have the same issue.

block6791, 2022-07-13:
I solved the issue. I set "enable updates" in the Brunch Framework options. I only set this option without actually upgrading. Then the Linux installation worked fine and is completely functional. I hope this helps others with this issue. This was done on Brunch + ChromeOS versions 102.

ankur-connect, 2022-07-13:
Same here...I did some hit and trial in Brunch configuration and it worked!!

SBShaown, 2022-09-09:
> I solved the issue. I set "enable updates" in the Brunch Framework options. I only set this option without actually upgrading. Then the Linux installation worked fine and is completely functional. I hope this helps others with this issue. This was done on Brunch + ChromeOS version 102.

I am having the same problem. Can you tell me how you fix that problem in more detail?