sebanc/brunch #1163: Change hardware ID

ekultails, 2021-07-23:
Installing Brunch appears to have modified the `HWID`/`hardware_class` as reported by `chrome://system`

Before (official Chrome OS 91): `VOXEL-GFMQ`
After (Brunch 91): `VOLTEER B2B-B2B-B2B-B2B-B2B-B66`

This is regarding this device:
- Name and model = Acer Chromebook Spin 713 (CP713-3W)
- Hardware ID = VOXEL-GFMQ
- Board = volteer

From what I have gathered, this seems to affect things such as which Chrome OS image updates will be provided.

The manual steps to change the hardware ID via the boot firmware does not work on Brunch.

https://www.reddit.com/r/chromeos/comments/embjdb/manually_updatechange_the_hwid/fdni6n8?utm_source=share&utm_medium=web2x&context=3

Is there a known workaround? Could this be something that can be added as a new option?

ekultails, 2021-07-23:
I figured it out! There is a script that is ran on boot to set the hardware ID.

```
chronos@localhost / $ cat /etc/init/hwid.conf 
start on starting boot-services

script
        echo "VOLTEER" > /sys/devices/platform/chromeos_acpi/HWID
end script
```

Following that same logic, setting the hardware ID can be done manually. After doing so, I was able to update to an image for an entirely different Chromebook!

```
chronos@localhost / $ sudo su -
localhost ~ # echo "VOXEL-GFMQ" > /sys/devices/platform/chromeos_acpi/HWID
```

Would love to still see a Brunch option added for this.

sebanc, 2021-07-23:
As far as I can tell there would be no benefit in doing this. Did it solve an issue or was it just an experiment ?

ekultails, 2021-07-24:
Yes, this did solve a few issues:

(1) None of the ChromeOS components (see `chrome://components`) were updated when I set `enable_updates` in Brunch and then went from Brunch 91 to Chrome OS 93. Most importantly, the `cros-termina` component did not update causing Crostini to no longer work after the update. In case you were wondering, no, this is not related to the recent DLC issue Chrome OS 91 had.

(2) I was able to update to an even newer image. The generic Volteer board on the Canary update channel is way behind the Acer Chromebook Spin 713 Canary and even Dev update channels. The platform version was lagging behind by many versions. By setting the hardware ID, I was able get an update to the newly released Chrome OS 94.

For more clarification, I am not actually using the Acer Chromebook Spin 713. I have been spoofing it on my auto-update expired Google Chromebook Pixel 2 to continue to get the latest cutting-edge updates.

Hope this helps to clarify the situation! I am a developer myself and have gone deep into learning of Chromium OS development over the past few months. This feature is something I could look into contributing on. Would love to get more involved with you and the Brunch project!

sebanc, 2021-07-24:
Ok so it seems to only be about the update process.

In that case, my suggestion would be to use a brunch bootscript to override the HWID with the one you want. To create the bootscript run:
```
sudo mkdir -p /var/brunch/bootscripts
echo 'echo "VOXEL-GFMQ" > /sys/devices/platform/chromeos_acpi/HWID' | sudo tee /var/brunch/bootscripts/hwid.sh
sudo chmod 0755 /var/brunch/bootscripts/hwid.sh
```

On another note, you indeed appear to know a lot about chromeos so do not hesitate to contribute. All contributions are greatly appreciated :)

ekultails, 2021-07-24:
Thanks for the tip and encouragement! I will go ahead and close this issue out.

Looking forward to finding more ways to contribute. With almost 500 GitHub issues open I am sure I can find something meaningful to work on. 😉

I would actually love to know more about the most important parts of Brunch. Perhaps that is a Discord conversation for another time. 🙂

closed 2021-07-24 by ekultails

sebanc, 2021-07-24:
Sure, PM me on Discord if you have questions about brunch internals