sebanc/brunch #1165: Play Store and its related apps is not opening

harshcans, 2021-07-24:
Sometimes Play Store and its related apps is not opening but running in background which creates a problem of freeze in my laptop. 

Thanks.

bm-a, 2021-07-24:
This is due to low disk speed.
I guess you are booting from HDD or a USB.
**Just wait on the login screen for about 2 to 3 minutes** _before typing your password_.
It'll fix your issue.

harshcans, 2021-07-24:
I have booted from USB

And I have person linked to chrome OS

bm-a, 2021-07-24:
Usb is too slow the android container crashes as it does not load when you log in.

Try shifting cros to Ssd or atleast a Hdd

harshcans, 2021-07-24:
Sorry It means that i have make bootable device for linux mint and then I have installed chrome os

bm-a, 2021-07-24:
_Just wait for some time(2-3 minutes) on the lock screen so that the android container fully loads._ **Unless it will crash**
 then play store and the android apps would hopefully work.

harshcans, 2021-07-25:
Thanks.
I hope its work.

harshcans, 2021-07-26:
If Play Store apps works and open then there is lack in my pc.
I cannot do any other work because of lack.

closed 2026-02-18 by github-actions[bot]