sebanc/brunch #1168: Chrome os black screen

sardine9284, 2021-07-27:
I successfully downloaded chrome os dual boot on my computer, but when i turn it on, it does not show anything and literally just goes black. I am using only uefi and also turned of secure boot. The only difference was that i used the linux mint grub customizer instead of grub2win. Please help, i really want to use chrome os.

kerta1n, 2021-08-31:
What's your setup? (Include everything from hardware to recovery image)

sardine9284, 2021-08-31:
> What's your setup? (Include everything from hardware to recovery image)

thx, i got that issue, just turned out i didn't copy one of the lines when copying the grub thingy? or what. but i have a new issue, when i boot chrome os, the loading brunch screen turns up as normal, but instead of showing the chrome os logo and letting me log in, it just goes back to the os selection screen. can you help me?