sebanc/brunch #1170: Is there a way to install Chrome OS brunch in VirtualBox/VMware?

cuj, 2021-07-27:
Hi, since brunch has actual Chrome OS, unlike Cloudready which has some apps removed, I was wondering, is there any method to install Brunch ChromeOS in VirtualBox/VMware or any guide to show how to install Brunch in VirtualBox/VMWare? Thanks.

WesBosch, 2021-08-13:
This is not currently supported, so there are no guides.

RJVB, 2021-08-19:
There are probably 2 approaches, both involving the use of the VM host's support for native disk access (for VirtualBox you'd use `VBoxManage internalcommands createrawvmdk -filename /path/to/diskimage.vmdk -rawdisk "your device"` ):
- Prepare the ChromeOS disk as described on real hardware, then create a VM using the corresponding .vmdk as boot disk
- Prepare a VM with a Linux boot image (you might even use a LiveCD image for that) and the ChromeOS disk .vmdk, boot it under Linux and then do the ChromeOS install in there.

Note however that ChromeOS might require certain drivers that are not available under a VM, or only after you install the GuestAdditions (and I have no idea how you'd do that). For comparison: I know Android-x86 will run n a VM but it will be very slow.

Jhone-Harry, 2022-11-13:
Hello, Installing ChromeOS on VMware Workstation Pro is what I want to do. There is a statement that "visual machines are not supported" on the GitHub downloads page. Is it possible to make work on VMware Workstation or VirtualBox?
Please, if you can assist, I appreciate it.

tejasraman, 2022-11-15:
Hi all!

As mentioned in #303 you can't really install ChromeOS in a VM due to a system Mesa that's incompatible with most SVGA cards (VMware style virtual cards). Could you please close this issue?

closed 2024-07-01 by cuj