sebanc/brunch #1172: Audio Not Working on AVITA Essentials Refresh : Cerleron N4020 (ALSA : No Soundcards Found)

nnilesh2, 2021-07-28:
Hi,
I have installed 'rammus 91 stable' image on my 'AVITA essential Refresh laptop : Celeron N4020', My with stable brunch_91.
Sound is not working on this after installation.
I have tried solutions like providing 'snd_hda_intel' in boot options.
[image]
[lsmod.txt](https://github.com/sebanc/brunch/files/6894200/lsmod.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/6894201/dmesg.txt)
[devices.txt](https://github.com/sebanc/brunch/files/6894202/devices.txt)

I am attaching dmesg.txt and lsmod.txt for reference. 

Thanks

sebanc, 2021-07-29:
Unfortunately I am not very good at debugging sound issues. Maybe you could try kernel 5.10 if you have not already tried ?

nnilesh2, 2021-07-29:
Thanks For your reply.
I already tried with Kernel-5.10 , still no luck !

sebanc, 2021-07-29:
Does sound work on other linux distros ? If it is the case, could you post a dmesg from a linux distro with working sound ?

nnilesh2, 2021-07-29:
Hi,
Sound and Wifi does not work on Linux_Min_20.1 Distro. (Earlier Wifi was not working on ChromeOS as well,I made wifi work on ChromeOS by passing 'rtl8821ce' in options )
Please find attached dmesg from Linux_Mint_20.1.
[dmesg_liux_mint_201.txt](https://github.com/sebanc/brunch/files/6898228/dmesg_liux_mint_201.txt)

nnilesh2, 2021-07-29:
Further update ,Audio Works fine on Windows 10, so sound card is in working condition.

sebanc, 2021-07-29:
Unfortunately your device sound card seems not supported in the linux kernel and there is not much I can do about it. You can create a bug on the linux kernel tracker.

nnilesh2, 2021-07-29:
Thank you for your help, I am closing issue.

closed 2021-07-29 by nnilesh2