sebanc/brunch #1173: Driver Intel WiFi 6E AX210

Twilighttony, 2021-07-29:
I upgraded to this wifi card and I noticed that in ChromeOS that this driver doesn't seem to exist. I have tried to enable the iwlwifi_backport flag as well as enabling kernel 5.10. On my OpenSUSE Tumbleweed install, this card works great. Any help would be awesome

sebanc, 2021-07-29:
In theory wifi should be supported, could you try to remove the file /lib/firmware/iwlwifi-ty-a0-gf-a0.pnvm ?

Source: https://www.reddit.com/r/Fedora/comments/m9p2h2/ax210_drivers/

MarcelKroner, 2021-08-17:
Good day sebanc,
i got the same problem like twilighttony. It's working with kernel-5.10 and your provided solution. You need to activate the iwlwifi option and right after you also need to delete the /lib/firmware/iwlwifi-ty-a0-gf-a0.pnvm file. Lastly reboot and the wifi adapter will be found. Later on i will test if it is also working on earlier kernel versions. 

Thank you very much for your great work! I hope you go on with this :)! You are awesome ^^

suleiman, 2022-02-14:
I don’t see any *.pnvm files in the lib/framework/ directory, they all end in .ucode 

Separately, if this fix requires Kernel 5.10 I’m out of luck as that kernel doesn’t boot on my machine, only kernel 5.4 does.

suleiman, 2022-02-14:
Loaded up brunch unstable r98 and am now able to boot into Chrome OS using Kernel 5.10! Thanks so much sebanc! 

My only issue now is the Wi-Fi not working with my AX210 wireless card. I’m frustratingly close. I just can’t find the pnvm files in the /lib/framework/ directory. All the files in there end in .ucode and I’ve already renamed the iwlwifi-ty-a0-gf-a0-59.ucode and iwlwifi-ty-a0-gf—a0-66.ucode files but no dice. 

Any ideas how to get this working?

Jejebond, 2022-10-22:
Hi, Did kernel 5.15 worked for you ? I can finally connect to wifi using ax210 chip but it won't work. I don't have any data transfer. Any idea ?

StellarOrbit, 2023-03-15:
I just reinstalled on a newer computer that has the same wifi card and the default 5.15 kernel works for me with wifi after I setup the device with ethernet, but 6.1 kernel breaks wifi again.