sebanc/brunch #1175: "amd-ucode.img not found" on a Intel CPU? (Chrome OS install on Legacy Bios)

Xerrious, 2021-07-29:
Hey guys!

I'm new here, this being my 1st post, with a weird problem. Weird at least for me:) I try since some days to install Chrome OS on Legacy Bios and when I thought I finally made it (at the point where it showed me in Linux Mint terminal that Chrome OS was successfully installed), after restart I got the error "file 'lib/firmware/amd-ucode.img' not found" and boot loop.

The thing is.. I have an Intel CPU, not AMD. And even so I still did the updates ( sudo apt-get install amd64-microcode etc.. I've tried pretty much everything I found internet about installing this ucode.. for both AMD and Intel.. but nothing works. I still get this frickin error.

 I'm not that advanced in Linux and stuff like that. But I is also not my 1st try with Chrome OS. I did it on a 7th gen UEFI, bootable USB, dual boot, etc.. but not on this laptop with Legacy Bios. And I really want to finish it, if it's not for this error. I use Brunch 83_k4.19 stable and samus 83 ( if I use latest version I get black screen). My CPU is Intel Core i3 M380 2.53GHz.

I would really appreciate an answer. Thank you very much!

WesBosch, 2021-08-13:
Those microcodes were not present in Brunch 83. it sounds like your grub entry is from a more recent release of brunch than the one you're using. You can either use a more recent release of Brunch (Brunch 91 is compatible with ChromeOS 83) or reinstall to get a valid grub entry

Xerrious, 2021-08-13:
Hey, thank you for reply. The thing is that I tried 91 and the same. That's why I went down till 83, coz I thought that it might work on an older version. So far I've tried 91, 85 and 83 and on all was the same problem

WesBosch, 2021-08-13:
if you switch to any version older than Brunch 85, make sure you're generating a new grub entry that does not include the ucode strings. this feature did not exist before Brunch 85, so the ucode error also did not exist on older releases.

Xerrious, 2021-08-13:
Thanks again. I will try it out and if it works I'll reply and close the thread

rkaradas, 2021-08-18:
I think the black screen is related to first gen Intel graphics [see notes](https://github.com/sebanc/brunch#hardware-support-and-added-features), so I guess you have to stick to 81.
If you are on 81, you have to edit the grub entry. You could

closed 2026-02-18 by github-actions[bot]