sebanc/brunch #1179: Android apps heating up Surface Laptop 3

Shadow0man, 2021-08-01:
I'm on hatch recovery on my SL3 and I've noticed that nearly any app especially with video playback is making laptop really hot and battery runs out very fast. For example Opera or TikTok, Disney+ app. It doesn't affect reddit for example. Is it kernel related or recovery? It doesn't seem right. In one game it got that much hot that game started to lag really badly after a while.

ZZZXXX0110, 2021-08-01:
You shouldn't be using Hatch, it has the new ARCVM and Android 11 container, instead of the current stable Android container with Android 9. The new ARCVM is not ready for release yet, and you just found out why.

Shadow0man, 2021-08-01:
I installed Volteer 90. It took me to use time machine page on recovery backup site unfortunately I got chrome CPU over 100 always bug in Volteer 90, the same was in Volteer 91. That's why I've installed Hatch in the first place. Is there a solution for this in Volteer? Or it is my hardware only related. It stops using over 100 only after a reboot, but it can go back to over 100 very quickly after reboot. Sometimes in a few minutes. Is there any other recovery I could use?