sebanc/brunch #1180: rtl8821ce support?

ticketsvl, 2021-08-01:
Hi.
My laptop wifi adapter rtl8821ce couldn`t start. 
As far as I understood, linux kernel 5.11 and 5.12 has this wifi adapter support [prof here](https://bugzilla.kernel.org/show_bug.cgi?id=210779)
Also, there is a unofficial dkms module [here](https://github.com/tomaspinho/rtl8821ce)
Is there any way to add this wifi in Chrome OS?
Best regards

zainullahk, 2021-08-11:
click ctrl + alt + t type shell enter then sudo su then click enter then type sudo edit-grub-config then in options= add rtl8821ce eg. options=rtl8821ce

zainullahk, 2021-08-11:
than click ctrl + alt + o click enter then ctrl + alt + x and then type reboot and click enter and reboot

closed 2026-02-18 by github-actions[bot]