sebanc/brunch #1185: SD card causing high CPU usage on surface go 2

The-Chosen-Sausage, 2021-08-04:
i have brunch installed on my surface go 2, using ramus recovery image. But the battery life wasn't great(10% in half an hour, normal browsing). My surface is also heating up. 
The external SD card seems to be causing the problem. According to the new diagnostic app, my CPU remains about 40%~50% usage with a 128GB SD card installed.  After i ejected the SD card, the usage drops to 10% immediately.

The SD card contains numerous small files from Windows. I've tried moving them into the hidden folder, or even using the .nomedia file to prevent Android media server from scanning it, but the result is the same.

[image]

I'm currently using the 4 GB variant(Intel 4425y, 64 GB emmc) with Windows and ChromeOS installed, which will be almost unusable without the SD card.

WesBosch, 2021-08-13:
this is pretty much the expected behavior due to the android subsystem's media scanner. I don't believe there's a workaround for avoiding this.

The-Chosen-Sausage, 2021-08-17:
I recently came up with an idea:
since Chrome OS is based on linux, we can actually set up some file access rule to deny the system from accessing it. Haven't tested it out yet, but I'll commet here if that worked.
Anyway thank you for replying!

closed 2022-04-11 by thechosensausage