sebanc/brunch #1192: How to dual boot windows from USB after installing Chrome OS

abdul4256, 2021-08-10:
Hi there,

I am novice at this so not quite sure what to do. I tried to go into recovery on my surface pro 4 after installing the chrome os via brunch and wanted to make a dual boot to add my windows as well but I keep getting grub and not entirely sure how to navigate it and install my windows 10 either as a dual boot or revert back entirely. Any help on this would be much appreciated. Thanks.

WesBosch, 2021-08-13:
If you want to dualboot with windows, you'll need to install windows first then brunch. If you do not currently have windows installed, you will need to create a windows USB (find a guide online, I'd suggest using another windows pc to make it)

Then to boot from that USB, refer to this for info.https://support.microsoft.com/en-us/surface/boot-surface-from-a-usb-device-fe7a7323-8d1d-823d-be17-9aec89c4f9f5

Afterwards, to dualboot with Brunch, make a seperate partition after windows (at least 15gb) and make a Brunch USB, then follow the dualboot with windows steps directed on the brunch github's readme file.

harshcans, 2021-08-17:
I can help you in booting Windows and Chrome OS with a live conversation and the issue will be sorted out frequently from Fiverr.
Buy this gig - https://www.fiverr.com/share/6VQjrR

WesBosch, 2021-08-17:
Absolutely not. If they want free support they can find us on discord, we don't allow advertising

harshcans, 2021-08-18:
Sorry It's my mistake