sebanc/brunch #1196: "Wrong EFI loader signature". After successfully getting into GNU GRUB section and selecting ChromeOS getting the "Wrong EFI loader signature" error.

NAS8910, 2021-08-11:
### Not able to boot into Chrome OS even after installation done by Linux terminal. 

> My System Configuration:
> 
> CPU: **Core 2 duo E7500**
> RAM: **4gb DD3**

My PC doesn't support UEFI. I tried by using tutorial with the mbr support material but couldn't boot into chromeOS. 

> _**Note:** I got the "ChromeOS installed" message in the Linux mint terminal after completing all the steps as per the guide._

I have attached a video and some photos which tells about bootloop going through with the EFI loader signature error.

[image]
[image]

_If you need any other information regarding this please inform_

_Sorry for the video to be faster than it was in real but couldn't get that under required size without it._

 

https://user-images.githubusercontent.com/80639377/129081854-d9cb0204-7f7a-482b-840e-bea18c06aef0.mp4

Ccl00101, 2021-08-13:
how long have you waited? first boot can take from 5 to 10 minutes.. if it boots normally after just ignore that wrong efi signature message.

NAS8910, 2021-08-13:
Actually When i try to boot into ChromeOS after the wrong efi screen the system reboots and this loops continues. You can refer the video to know the actual scene for the system.

WesBosch, 2021-08-13:
Your CPU is not supported, and will not boot Brunch regardless of what settings you change. Please see the corresponding section of the readme or this page for further info. https://github.com/sebanc/brunch/wiki/CPUs-&-Recoveries

NAS8910, 2021-08-13:
Oh okay that the reason. One last question Can i install neverware's version of ChromeOS or not. If not which OS should i opt then?
Thank you for your responses.

WesBosch, 2021-08-13:
Neverware is also dropping support for Core 2 Duo devices soon. I'd honestly suggest updating, a 12+ year old cpu is going to lose support from a lot of distros.

There are some lightweight Linux distros you could try, but there are so many I couldn't suggest just one.

NAS8910, 2021-08-13:
Thank you so much to all.

closed 2021-08-13 by NAS8910